Koenkk/Z-Stack-firmware #98: Impossible to unzip all firmware

vrolland78, 2019-06-24:
Hi,
I hope this is not a stupid question, but with all the tools I try on Windows 7/10 (winrar, 7zip) or linux command line (unzip) I can not extract the firmware.

I have this kind of error

Archive:  CC2531_20190425.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of CC2531_20190425.zip or
        CC2531_20190425.zip.zip, and cannot find CC2531_20190425.zip.ZIP, period.

Is there any issue with the files or is it just me ?
Thaks for your help and your work.
Vincent

Xeroxxx, 2019-06-24:
Just don't do right click and Save as...

Click on the zipfile and Download up right.

vrolland78, 2019-06-24:
f..k ! It was so simple :(

closed 2019-06-24 by vrolland78