Koenkk/Z-Stack-firmware #101: Note about current version: CC2531_DEFAULT_20190608.zip

DerElch87, 2019-07-09:
Hi,
this is only my personal experience, but probably it can help for future versions or whatever.

For me the latest version: CC2531_DEFAULT_20190608.zip is so far the most unstable one.
In my network (17 devices with 5 routers including) I have a lot of problems with not or late reacting bulbs etc.
Sometimes I have to switch buttons many times (with about 10s waiting between) or sometimes bulbs not react at all.

I will try the max stability version today and hope that this one will work better.

Best regards.

Koenkk, 2019-07-09:
I would recommend https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing in that case

DerElch87, 2019-07-17:
The source routing firmware is now running since 8 days.
No issues so far, running perfectly fine.
Thank you!

DerElch87, 2019-07-17:
And of cource,
keep on doing the good work!
Really appreciate it!

closed 2019-07-17 by DerElch87