Koenkk/Z-Stack-firmware #104: CC2531 router red led blinking, no connection (and: stopped pairing devices all together)

prankousky, 2019-08-11:
Hi everybody,

I have been using zigbee for a long time, but just recently switched to `zigbee2mqtt`. I have multiple sensors set up successfully, and they currently do report via mqtt (so the coordinator must be running without any issues I assume).

* coordinator `CC2531`
* router 1 `CC2530` 2530 std firmware
* router 2 `CC2530` 2530 diag firmware
* router 3 **not working** `CC2531` (tried both 2531 diag and 2531 std firmware)

As soon as I connect what I refer here to as `router 3`, it starts flashing a red led. If I hold `S2` for >5 seconds, flashing stops for a second or two, then starts again.

If I press the other button, a green led stays lid next to the red one. If I press it again, the green led disappears again. Hold `S2` will still not work.

My first attempt was mit diag firmware, then I switched to std. I don't quite understand this. I flashed the coordinator myself as well as both other routers. Everything worked fine. Flashing the `CC2531` is even simpler as you just have to connect the debugger and place the cable on the pins according to documentation (which I did each time). `cc-tool` output always displayed success, which it would not if device not connect or connected not properly.

Just now I tried two things to test my coordinator:

1. test a xiami button, results `{"linkquality":68,"battery":100,"voltage":3032,"click":"single"}`
==> coordinator is working, as it picks up the single button click
2. try to connect a xiaomi temperature sensor
==> no `device incoming`, device not connecting, no status updates from device, no new device added to bottom of `configuration.yaml`

I then `docker stop zigbee2mqtt`, unplugged the coordinator, plugged it back in, `docker run zigbee2mqtt`. Started this  test over from beginning. It will again pick up the button press, but not connect the temperature sensor, nor the `CC2531 router`.

My configuration.yaml starts with

```
homeassistant: true
permit_join: true
(...)
```

So joining **should** work. I just realized that joining does not only *not* work for the new router, but also not for any other device (or at least not for that temperature sensor I tested; I inserted a new battery into this sensor to see if that would change anything, which it did not).

Last time I've had a problem with `zigbee2mqtt`, I deleted **everything** in the config folder (including `database.db` and `.storage`). However, if I did that now, I would have to re-pair **every single** device in my house. That's 3 devices per room (door, window, temperature), plus whatever else I might have (buttons, xiaomi cube, hue lights, ...). So if there is any way to avoid starting with a clean config, I'd very much prefer this.

I just went through the log, here are some perhaps relevant details:

```
Coordinator firmware version: 20190608
Currently 44 devices are joined
Device_Router_01 (...): CC2530.ROUTER - Custom devices (DiY) [CC2530 router) (Router)
(...)
Device_Router_02 (...): CC2530.ROUTER - Custom devices (DiY) [CC2530 router) (Router)
`permit_join_ set to `true` in configuration.yaml.
Allowing new devices to join.
zigbee-shepherd ready
Connnected to MQTT server
```

I `grep`ped the most recent log for `incoming` and `join`, but there was no result for incoming and only regular debugging (no attempts nor errors) for join.

I was less than 30cm away from the coordinator when attempting to pair both the xiaomi sensors and the CC2531 router. I have a router in my room as well, which I unplugged to test whether or not it might interfere.

The coordinator is a `CC2531` USB stick, but it has an external antenna, which is on a cable about 1 meter away from my server. So I assume there is no interference from the device the stick is connected to. I also had a USB extension connected between the USB stick and my device, but since I did not notice any difference, I removed the extension and plugged it back right into the USB slot of my server).

What are possible ways to fix this without having to re-pair all my already connected devices? 

Thanks in advance for your support :)

prankousky, 2019-08-20:
Anybody got any idea how I might be able to fix this?

I have tried to stop the docker container, unplug the coordinator, plug it back in, restart the docker container. Nothing. Due to using multiple routers (and devices such as Phillips Hue light bulbs, which should also act like routers, correct?) I should be able to pair more than those 44 paired devices. But it does not work. 

I am very hesitant to re-flashing the coordinator once again because that would mean I'd have to re-pair each of those 44 devices again, correct? Or can I re-flash it, then feed it  the same configuration file as before (that already has each device's ID in it) and they will automatically be paired again (as the coordinator was the same as before)? IF so, I would gladly give it a shot, but risking having to re-pair all devices without knowing whether or not this issue would be fixed then is too much of a risk for me (just because last time it took me half a day to re-pair all devices - including gathering them and putting them back where they belong).

Thanks for your help.

ptvoinfo, 2019-08-20:
@prankousky 44 devices? I assume you are over the limit on your coordinator or router. Therefore the new device cannot connect (no free slots).

prankousky, 2019-08-20:
@ptvoinfo thank you. I thought of that, too. But see [here](https://www.zigbee2mqtt.io/information/zigbee_network.html) ==>

> The default CC2531 USB sniffer coordinator firmware has a limit of 20 direct children. This means 1 coordinator + 20 end devices and 0 routers.
However by adding routers to your network you can overcome the limit of 20 devices. This means that, e.g. a network of 1 coordinator, 4 routers and 50 end devices is possible.
The number of childs that a router support differs per device! Therefore, the increased size of a network by adding a router is arbitrary.

If I have 44 devices already, the routers **must** work as intended, because otherwise I wouldn't have been able to pair more than 20 devices. So even if my two routers are *almost* at their maximum devices level, I should be able to connect 6 more devices. I am, by the way, only trying to add this new router to have the ability of pairing more devices (not to improve the range). But unfortunately it does not work.

And I am not completely sure about this, but afaik, things like Hue bulbs act as routers **and** regular devices, so having multiple Hue bulbs should allow me to add even more devices..?

ptvoinfo, 2019-08-20:
When you try to connect a new device it may try to connect to the coordinator directly, but there are no free slots.

prankousky, 2019-08-20:
I believe I've found the problem: one of my Hue Bulbs was paired, but not correctly. It would sometimes execute commands (toggle on/off), and sometimes not. I removed it from my configuration, then re-paired it. Now I was also able to pair both my `CC2531` router, as well as another `CC2530` that I just flashed to test whether or not this was caused by the CC2531 router.

Can I specify an order in which devices should try? FOr example, if they try the coordinator but it doesn't have any slots left, try each router, starting with the one that has most free slots? Or similar?

ptvoinfo, 2019-08-20:
No, you cannot specify an order. I assume end devices select a parent node by link quality and path length (nodes between the device and the coordinator). If link quality is good, but a path is too long, the device may select anything (I cannot answer precisely, in Zstack, it is hidden in compiled code).

prankousky, 2019-08-20:
Thank you @ptvoinfo  :)

I guess I'll leave it at this for now, then.  So far, everything works fine, and it seems like the error was caused by the "broken" Philips Hue pairing.

closed 2019-08-20 by prankousky