Koenkk/Z-Stack-firmware #105: Firmware 20190608 (default) - stick hangs

DubhAd, 2019-08-17:
Running Z2M 1.5.1, with a CC2531 on the 20190608 (Default) firmware, I'm running into an odd problem. Every so often (5 days to a couple of weeks) the stick stops responding, everything goes offline, and I have to unplug and reconnect it, then restart Z2M.

I didn't see this problem on the previous (20190425) firmware, though I did have stability issues.

It _looks_ to be a firmware issue, but I'm not finding anybody else reporting the problem, and there's no newer firmware to upgrade to.

Koenkk, 2019-08-18:
Could you try with: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing (doesn't need repairing).

DubhAd, 2019-08-18:
Thanks. I'll give that a shot

DubhAd, 2019-08-18:
Flashed, did result in a few devices (most annoying being the Gledopto LED controller) dropping off the mesh until I moved some Ikea routers around. I'll update with progress.

DubhAd, 2019-08-21:
Nope, didn't help :(

Koenkk, 2019-08-22:
Do you use one of the following features? 
- Availability
- Reporting
- Networkmap

DubhAd, 2019-08-22:
I do use NetworkMap and Availability, but not Reporting.

I can disable any or all of those if it'll help stability.

Koenkk, 2019-08-23:
Yes please try that. (especially the networkmap).

DubhAd, 2019-08-23:
I disabled them both shortly after I'd posted, so we'll see how it goes.

Thanks

For reference, my config is now:

```yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.2.30'
  user: some_user
  password: some_secret
serial:
  port: /dev/ttyACM0
advanced:
  pan_id: 0xdead
  ext_pan_id: [0xde, 0xad, 0xbe, 0xef, 0xca, 0xfe, 0x01, 0x02]
  channel: 25
  cache_state: true
  network_key: [0xde, 0xad, 0xbe, 0xef, 0xca, 0xfe, 0x01, 0x02, 0xde, 0xad, 0xbe, 0xef, 0xca, 0xfe, 0x01, 0x02]
  last_seen: 'ISO_8601'
  report: false
devices: devices.yaml
groups: groups.yaml
device_options:
  occupancy_timeout: 70
  humidity_precision: 1
  temperature_precision: 1
  retain: true
```

DubhAd, 2019-08-29:
Well, that _looks_ to have solved the hard hang of the stick. I now get a soft hang instead 😃 

The first eight lines in this snippet are while it's all working, from then on it's non-responsive. This has happened twice now:

First:
```
8/22/2019, 6:11:05 PM - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-08-22.18-11-04'
8/22/2019, 6:11:05 PM - info: Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
8/22/2019, 6:11:05 PM - info: Starting zigbee-shepherd
...
8/25/2019, 7:11:08 PM - warn: Device with modelID 'undefined' is not supported.
8/25/2019, 7:11:08 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
8/25/2019, 7:11:30 PM - info: MQTT publish: topic 'zigbee2mqtt/router1', payload '{"state":true,"linkquality":55,"last_seen":"2019-08-25T18:11:30.017Z","description":"12719/0x00158D00033B752A","type":"CHILD_RFD","rssi":170,"led_state":true}'
8/25/2019, 7:11:30 PM - info: MQTT publish: topic 'zigbee2mqtt/router1', payload '{"state":true,"linkquality":55,"last_seen":"2019-08-25T18:11:30.114Z","description":"0/0x00124B0018E1F2ED","type":"COORD","rssi":34,"led_state":true}'
--- fails here ---
8/25/2019, 7:12:11 PM - error: Failed to ping '0x00124b0018e1f6a4'
8/25/2019, 7:13:11 PM - error: Failed to ping '0x00124b0018e1f6a4'
8/25/2019, 7:14:11 PM - error: Failed to ping '0x00124b0018e1f6a4'
8/25/2019, 7:14:43 PM - info: Zigbee publish to device 'downstairs_hall', genLevelCtrl - moveToLevelWithOnOff - {"level":10,"transtime":165} - {"manufSpec":0,"disDefaultRsp":0} - 11
8/25/2019, 7:14:46 PM - error: Zigbee publish to device 'downstairs_hall', genLevelCtrl - moveToLevelWithOnOff - {"level":10,"transtime":165} - {"manufSpec":0,"disDefaultRsp":0} - 11 failed with error Error: request timeout
8/25/2019, 7:14:46 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: request timeout","meta":{"entity":{"ID":"0x00124b001ba70aff","type":"device","friendlyName":"downstairs_hall"},"message":"{\"state\": \"ON\", \"transition\": 5, \"brightness\": 10, \"color_temp\": 291}"}}'
8/25/2019, 7:14:46 PM - info: Zigbee publish to device 'downstairs_hall', lightingColorCtrl - moveToColorTemp - {"colortemp":291,"transtime":544.5} - {"manufSpec":0,"disDefaultRsp":0} - 11
8/25/2019, 7:14:49 PM - error: Zigbee publish to device 'downstairs_hall', lightingColorCtrl - moveToColorTemp - {"colortemp":291,"transtime":544.5} - {"manufSpec":0,"disDefaultRsp":0} - 11 failed with error Error: request timeout
8/25/2019, 7:14:49 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: request timeout","meta":{"entity":{"ID":"0x00124b001ba70aff","type":"device","friendlyName":"downstairs_hall"},"message":"{\"state\": \"ON\", \"transition\": 5, \"brightness\": 10, \"color_temp\": 291}"}}'
8/25/2019, 7:15:11 PM - error: Failed to ping '0x00124b0018e1f6a4'
8/25/2019, 7:16:11 PM - error: Failed to ping '0x00124b0018e1f6a4'
8/25/2019, 7:17:11 PM - error: Failed to ping '0x00124b0018e1f6a4'
8/25/2019, 7:18:11 PM - error: Failed to ping '0x00124b0018e1f6a4'
```

Second:
```
8/25/2019, 9:19:14 PM - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-08-25.21-19-13'
8/25/2019, 9:19:18 PM - info: Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
...
8/29/2019, 9:50:51 AM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
8/29/2019, 9:51:21 AM - warn: Device with modelID 'undefined' is not supported.
8/29/2019, 9:51:21 AM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
8/29/2019, 9:51:50 AM - info: MQTT publish: topic 'zigbee2mqtt/router1', payload '{"state":true,"linkquality":63,"last_seen":"2019-08-29T08:51:50.183Z","description":"53682/0x00158D00033B74C1","type":"CHILD_RFD","rssi":170,"led_state":true}'
--- fails here ---
8/29/2019, 9:51:53 AM - error: Failed to ping 'family_room_router'
8/29/2019, 9:52:54 AM - error: Failed to ping 'family_room_router'
8/29/2019, 9:53:53 AM - error: Failed to ping 'family_room_router'
8/29/2019, 9:54:54 AM - error: Failed to ping 'family_room_router'
```

It carries on like that until Z2M is restarted, but it does restart and work without issue. That I can at least monitor and automate a restart.

Koenkk, 2019-08-30:
This should be checked again after https://github.com/koenkk/zigbee2mqtt/issues/1888

closed 2020-08-18 by github-actions[bot]