Koenkk/Z-Stack-firmware #109: CC2531 router buttons

muxa, 2019-09-15:
I was testing the router, it had the LED flashing red every few seconds. 
Pressing S1 button would toggled the green LED. 
I pressed and help the S1 button for 6 seconds, and LED is now off and not flashing. And the router does not seem to be joining the network. 
So my questions are:

1) What did i do by holding the S1 button for?
2) How do i make the router work again?
3) What do S1 and S2 buttons do on a CC2531 with a router firmware?

ptvoinfo, 2019-09-15:
Hi @muxa,

The S1 button only changes the state of the green LED. Try to re-pair the router using a button or a power on/off procedure.

Lights
=========================

Short fast blinks (one per second) – the router is connecting to a network.
Short long blinks (one per 4 seconds) – normal operations.
Three short blinks – the router cannot send a report to a coordinator.

Pairing
=========================
Flash firmware and permit joining to a network on your coordinator.

Re-pairing
=========================
CC2530, CC2531: Power on, wait 2 seconds, power off, repeat this cycle three times.
CC2531: Power on, press and hold down the S2 button for 5 seconds.

tb-killa, 2019-10-11:
@ptvinfo don't know if this question come up again in the past but does we could toggle led state of router by zigbee message? I see led state as attribute by reporting value if pushing button on pcb.

closed 2019-11-12 by muxa