Koenkk/Z-Stack-firmware #113: Confusing downloads

bochelork, 2019-09-28:
Default firmware for coordinator say:

`CC2531_DEFAULT_20190608.zip	Source routing firmware. 	4 months ago`

At: Z-Stack-firmware/coordinator/Z-Stack_Home_1.2/bin/default/

It's somewhat confusing...

sdv22, 2020-01-16:
Seems it has SR binary cause my dongle flashed with it is limited to 5 direct children.
@Koenkk , Could you please verify it? I do not have any other way to flash except SBL.

Koenkk, 2020-01-16:
What do i need to verify? SBL or hex firmware is identical except the flashing method.

sdv22, 2020-01-16:
> 
> 
> What do i need to verify? SBL or hex firmware is identical except the flashing method.

I compiled it from sources. Seems binary here is correct. But the problem is that it doesn't connect more than 5 enddevices. The problem arised after I flashed sr firmware and then returned back to non-sr. Maybe I need to reset the stick.

closed 2020-08-18 by github-actions[bot]