Koenkk/Z-Stack-firmware #115: Credit for firmware ?

ngodinhduy, 2019-10-02:
how can i credit for your firmware
_https://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/_

Koenkk, 2019-10-02:
@ptvoinfo please check

ptvoinfo, 2019-10-02:
@Koenkk 'Thank you' is enough :)

ngodinhduy, 2019-10-02:
I really want your source code to adjust to UART output, and work with another MCU.

ptvoinfo, 2019-10-02:
The router firmware contains nothing special. You may download ZStack and find several examples. I've used it too.

ngodinhduy, 2019-10-02:
I tried, but I always got this error even though I was running in router mode, it didn't mesh so the distance was very low.
[image]

ptvoinfo, 2019-10-02:
End devices cannot provide LQI information.

ngodinhduy, 2019-10-03:
Yes, but i  build it like Router
[image]

ngodinhduy, 2019-10-03:
[image]

closed 2020-08-18 by github-actions[bot]