Koenkk/Z-Stack-firmware #118: Some Questions about cc1352p

flower87lucky, 2019-10-14:
I bought a CC1352P chip (http://www.ebyte.com), and I wanted to connect it to the raspberry pie through the UART port. Can you use this firmware directly?

Koenkk, 2019-10-14:
I expect it can be connected through UART but i'm not sure.

dzungpv, 2020-01-30:
@flower87lucky late reply but now i am wiring it directly to PI,
You can following the step below:
1, Uninstall all jumper: GND, 3.3V, 5V, TMS, TCK, TX, RX. Later if you want to flash it, just reinstall.
2, Change the power Jumper from XDS110 power position to External power
3, Wire Pi 3.3v to 3.3v pin, GND-> GND, Pi RX>IO13, Pi TX->IO12
4, Disable Bluetooth driver on Pi
@Koenkk You can add this to documents

dzungpv, 2020-01-30:
@flower87lucky You can use it, flash the firmware with https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P_2_20191106.zip, or you can build your own
wire PI RX-> DIO13, PI TX->DIO12, PI 3.3 to 3.3V, GND to GND

closed 2020-08-18 by github-actions[bot]