Koenkk/Z-Stack-firmware #119: How to set up when used cc1352p-2

flower87lucky, 2019-10-19:
"serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 6758,
    "channel": 15,

Starting zigbee-herdsman...
zigbee2mqtt:error 2019-10-19T14:13:22: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-10-19T14:13:22: Failed to start zigbee
zigbee2mqtt:error 2019-10-19T14:13:22: Exiting...
zigbee2mqtt:error 2019-10-19T14:13:22: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm
 ERR! 
code ELIFECYCLE
npm
 ERR!
 errno 1
npm
 ERR! zigbee2mqtt@1.6.0 start: `node index.js`
npm
 ERR! Exit status 1
npm ERR!
 
npm ERR!
 Failed at the zigbee2mqtt@1.6.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR!
 A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-10-19T14_13_22_498Z-debug.log
2019-10-19T22:13:22: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-10-19T22:13:22: PM2 log: App [npm:0] starting in -fork mode-
2019-10-19T22:13:22: PM2 log: App

a-bailey, 2019-11-14:
I'm having the same issue, also I get failures on both write tests on the ports.
I currently use cc1352p-2 HW Rev: B could that be the problem? I'm not that deep into RF Hardware and can't really debug the problem.

Also tried "serial": {
"port": null
},

didnt make a difference

a-bailey, 2019-11-14:
[image]

Tam-Lin, 2019-11-16:
I'm having the same issue.

Koenkk, 2019-11-17:
@frederikg should revision B work with the current firmware?

Optikk, 2019-11-19:
Have the same issue with rev B

dreimer1986, 2019-11-19:
Confirmed, same problem here... Ti sent me that one (via Mouser to be precise) so Rev B is not supported? Maybe that should be added to the Wiki then to prevent these problems. Or is a support planned/possible?

Koenkk, 2019-11-19:
I just checked and mine is also rev B, so it should work, make sure that you are on zigbee2mqtt 1.7 and the port is correct

dreimer1986, 2019-11-19:
zigbee2mqtt version 1.7.1+dev (commit #ba18c5f773c160b54977908ed241cd40547a2fff)

I have:
/dev/ttyACM0
/dev/ttyACM1
added by the device. ACM1 causes a timeout error alomost immediately. ACM0 takes about 1 minute to timeout.

ACM0:
´´´
Debug information
-----
-----
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": true,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://homeassistant",
    "user": "xxxxx",
    "password": "xxxxx",
    "include_device_information": true
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blacklist": [],
    "report": true,
    "soft_reset_timeout": 60,
    "last_seen": "ISO_8601"
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1.0,
    "initialdelay": 1.0,
    "options": "-d -d",
    "log": false
  }
}Debug information end
./run.sh: line 24: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2019-11-19T23:50:10: PM2 log: Launching in no daemon mode
2019-11-19T23:50:10: PM2 log: App [npm:0] starting in -fork mode-
2019-11-19T23:50:10: PM2 log: App [npm:0] online
> zigbee2mqtt@1.7.1 start /app
> node index.js
zigbee2mqtt:info  2019-11-19 23:50:12: Logging to directory: '/share/zigbee2mqtt/log/2019-11-19.23-50-11'
zigbee2mqtt:info  2019-11-19 23:50:12: Starting zigbee2mqtt version 1.7.1+dev (commit #ba18c5f773c160b54977908ed241cd40547a2fff)
zigbee2mqtt:info  2019-11-19 23:50:12: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-11-19 23:51:17: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-11-19 23:51:17: Failed to start zigbee
zigbee2mqtt:error 2019-11-19 23:51:17: Exiting...
zigbee2mqtt:error 2019-11-19 23:51:17: Error: AREQ - ZDO - stateChangeInd after 60000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR!
 zigbee2mqtt@1.7.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
npm 
ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-11-19T22_51_17_675Z-debug.log
´´´

ACM1:
´´´
2019-11-19T23:53:58: PM2 log: Launching in no daemon mode
2019-11-19T23:53:58: PM2 log: App [npm:0] starting in -fork mode-
2019-11-19T23:53:58: PM2 log: App [npm:0] online
> zigbee2mqtt@1.7.1 start /app
> node index.js
zigbee2mqtt:info  2019-11-19 23:53:59: Logging to directory: '/share/zigbee2mqtt/log/2019-11-19.23-53-59'
zigbee2mqtt:info  2019-11-19 23:53:59: Starting zigbee2mqtt version 1.7.1+dev (commit #ba18c5f773c160b54977908ed241cd40547a2fff)
zigbee2mqtt:info  2019-11-19 23:53:59: Starting zigbee-herdsman...
zigbee2mqtt:error 2019-11-19 23:54:07: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-11-19 23:54:07: Failed to start zigbee
zigbee2mqtt:error 2019-11-19 23:54:07: Exiting...
zigbee2mqtt:error 2019-11-19 23:54:07: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.7.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-11-19T22_54_07_144Z-debug.log
´´´

Koenkk, 2019-11-20:
@dreimer1986 `/dev/ttyACM0` is the correct port, changing the pan_id in configuration.yaml will fix this issue (the default pan id: 0x162a is already used by your old network), try `0x163a`.

dreimer1986, 2019-11-20:
Will try that evening at home. Before I forget. Uniflash did a Firmware update on full wipe, so this is newer than stock in my case too. Just to have told about that. Will tell my results later

Optikk, 2019-11-20:
Hi. I did some more tests and what I have found.

My config:
`{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": true,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://192.168.1.22:1883",
    "user": "mqtt_admin",
    "password": "password"
  },
  "serial": {
    "port": "/dev/ttyACM0"
  },
  "advanced": {
    "pan_id": 5690,
    "log_level": "debug",
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "rtscts": true,
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1,
    "initialdelay": 1,
    "options": "-d -d",
    "log": false
  }
}`

I'm using z2m 1.7.1 hassio addon. When I plug coordinator board and start addon it starts normally.

`[Info] Socat is DISABLED and not started
2019-11-20T17:06:27: PM2 log: Launching in no daemon mode
2019-11-20T17:06:27: PM2 log: App [npm:0] starting in -fork mode-
2019-11-20T17:06:27: PM2 log: App [npm:0] online
> zigbee2mqtt@1.7.1 start /zigbee2mqtt-1.7.1
> node index.js
zigbee2mqtt:info  2019-11-20T17:06:28: Logging to directory: '/share/zigbee2mqtt/log/2019-11-20.17-06-27'
zigbee2mqtt:debug 2019-11-20T17:06:28: Removing old log directory '/share/zigbee2mqtt/log/2019-11-20.17-03-17'
zigbee2mqtt:debug 2019-11-20T17:06:28: Loaded state from file /share/zigbee2mqtt/state.json
zigbee2mqtt:info  2019-11-20T17:06:28: Starting zigbee2mqtt version 1.7.1 (commit #unknown)
zigbee2mqtt:info  2019-11-20T17:06:28: Starting zigbee-herdsman...
zigbee2mqtt:debug 2019-11-20T17:06:28: Using zigbee-herdsman with settings: '{"network":{"panID":5690,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
zigbee2mqtt:info  2019-11-20T17:06:32: zigbee-herdsman started
zigbee2mqtt:info  2019-11-20T17:06:32: Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'
zigbee2mqtt:debug 2019-11-20T17:06:32: Zigbee network parameters: {"panID":5690,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:info  2019-11-20T17:06:32: Currently 1 devices are joined:
zigbee2mqtt:info  2019-11-20T17:06:32: 0x00158d00026759fb (0x00158d00026759fb): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
zigbee2mqtt:warn  2019-11-20T17:06:32: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2019-11-20T17:06:32: Allowing new devices to join.
zigbee2mqtt:warn  2019-11-20T17:06:32: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2019-11-20T17:06:32: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2019-11-20T17:06:32: Connecting to MQTT server at mqtt://192.168.1.22:1883
zigbee2mqtt:info  2019-11-20T17:06:32: Connected to MQTT server`

But if I stop addon and start it again then I have an error:
`[Info] Socat is DISABLED and not started
2019-11-20T17:10:34: PM2 log: Launching in no daemon mode
2019-11-20T17:10:34: PM2 log: App [npm:0] starting in -fork mode-
2019-11-20T17:10:34: PM2 log: App [npm:0] online
> zigbee2mqtt@1.7.1 start /zigbee2mqtt-1.7.1
> node index.js
zigbee2mqtt:info  2019-11-20T17:10:35: Logging to directory: '/share/zigbee2mqtt/log/2019-11-20.17-10-34'
zigbee2mqtt:debug 2019-11-20T17:10:35: Removing old log directory '/share/zigbee2mqtt/log/2019-11-20.17-03-26'
zigbee2mqtt:debug 2019-11-20T17:10:35: Loaded state from file /share/zigbee2mqtt/state.json
zigbee2mqtt:info  2019-11-20T17:10:35: Starting zigbee2mqtt version 1.7.1 (commit #unknown)
zigbee2mqtt:info  2019-11-20T17:10:35: Starting zigbee-herdsman...
zigbee2mqtt:debug 2019-11-20T17:10:35: Using zigbee-herdsman with settings: '{"network":{"panID":5690,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
zigbee2mqtt:error 2019-11-20T17:10:42: Error while starting zigbee-herdsman
zigbee2mqtt:error 2019-11-20T17:10:42: Failed to start zigbee
zigbee2mqtt:error 2019-11-20T17:10:42: Exiting...
zigbee2mqtt:error 2019-11-20T17:10:42: Error: SRSP - SYS - version after 6000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/zigbee2mqtt-1.7.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR!
 zigbee2mqtt@1.7.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
npm
 ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm
 ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-11-20T17_10_42_432Z-debug.log
2019-11-20T17:10:43: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-11-20T17:10:43: PM2 log: App [npm:0] starting in -fork mode-`

But if I stop addon, unplug board and plug it again and then start addon then averything is ok.

What it can be?

Koenkk, 2019-11-20:
What is shown in the log when you stop the addon? Does it stop without errors?

Optikk, 2019-11-20:
After stoping addon in logs nothing:

`info  2019-11-20T18:39:21: Logging to directory: '/share/zigbee2mqtt/log/2019-11-20.18-39-20'
debug 2019-11-20T18:39:21: Loaded state from file /share/zigbee2mqtt/state.json
info  2019-11-20T18:39:21: Starting zigbee2mqtt version 1.7.1 (commit #unknown)
info  2019-11-20T18:39:21: Starting zigbee-herdsman...
debug 2019-11-20T18:39:21: Using zigbee-herdsman with settings: '{"network":{"panID":5690,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"}}'
info  2019-11-20T18:39:25: zigbee-herdsman started
info  2019-11-20T18:39:25: Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'
debug 2019-11-20T18:39:25: Zigbee network parameters: {"panID":5690,"extendedPanID":"0xdddddddddddddddd","channel":11}
info  2019-11-20T18:39:25: Currently 1 devices are joined:
info  2019-11-20T18:39:25: 0x00158d00026759fb (0x00158d00026759fb): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
warn  2019-11-20T18:39:25: `permit_join` set to  `true` in configuration.yaml.
warn  2019-11-20T18:39:25: Allowing new devices to join.
warn  2019-11-20T18:39:25: Set `permit_join` to `false` once you joined all devices.
info  2019-11-20T18:39:26: Zigbee: allowing new devices to join.
info  2019-11-20T18:39:26: Connecting to MQTT server at mqtt://192.168.1.22:1883
info  2019-11-20T18:39:26: Connected to MQTT server
info  2019-11-20T18:39:26: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2019-11-20T18:39:26: MQTT publish: topic 'zigbee2mqtt/0x00158d00026759fb', payload '{"state":"ON","power":0,"voltage":220.5,"consumption":0.01,"temperature":40,"linkquality":102}'
info  2019-11-20T18:39:26: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.7.1","commit":"unknown","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}},"log_level":"debug","permit_join":true}'
info  2019-11-20T18:39:26: MQTT publish: topic 'homeassistant/switch/0x00158d00026759fb/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x00158d00026759fb/set","state_topic":"zigbee2mqtt/0x00158d00026759fb","name":"0x00158d00026759fb_switch","unique_id":"0x00158d00026759fb_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00026759fb"],"name":"0x00158d00026759fb","sw_version":"Zigbee2mqtt 1.7.1","model":"Mi power plug ZigBee (ZNCZ02LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-20T18:39:26: MQTT publish: topic 'homeassistant/sensor/0x00158d00026759fb/power/config', payload '{"unit_of_measurement":"W","icon":"mdi:factory","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/0x00158d00026759fb","json_attributes_topic":"zigbee2mqtt/0x00158d00026759fb","name":"0x00158d00026759fb_power","unique_id":"0x00158d00026759fb_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00026759fb"],"name":"0x00158d00026759fb","sw_version":"Zigbee2mqtt 1.7.1","model":"Mi power plug ZigBee (ZNCZ02LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2019-11-20T18:39:26: MQTT publish: topic 'homeassistant/sensor/0x00158d00026759fb/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d00026759fb","json_attributes_topic":"zigbee2mqtt/0x00158d00026759fb","name":"0x00158d00026759fb_linkquality","unique_id":"0x00158d00026759fb_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d00026759fb"],"name":"0x00158d00026759fb","sw_version":"Zigbee2mqtt 1.7.1","model":"Mi power plug ZigBee (ZNCZ02LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
debug 2019-11-20T18:39:26: Received Zigbee message from '0x00158d00026759fb', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":0}' from endpoint 2 with groupID 0
info  2019-11-20T18:39:26: MQTT publish: topic 'zigbee2mqtt/0x00158d00026759fb', payload '{"state":"ON","power":0,"voltage":220.5,"consumption":0.01,"temperature":40,"linkquality":99}'
debug 2019-11-20T18:39:42: Received Zigbee message from '0x00158d00026759fb', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":0}' from endpoint 2 with groupID 0
info  2019-11-20T18:39:42: MQTT publish: topic 'zigbee2mqtt/0x00158d00026759fb', payload '{"state":"ON","power":0,"voltage":220.5,"consumption":0.01,"temperature":40,"linkquality":102}'
debug 2019-11-20T18:39:58: Received Zigbee message from '0x00158d00026759fb', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":0}' from endpoint 2 with groupID 0
info  2019-11-20T18:39:58: MQTT publish: topic 'zigbee2mqtt/0x00158d00026759fb', payload '{"state":"ON","power":0,"voltage":220.5,"consumption":0.01,"temperature":40,"linkquality":105}'
`

and thats all. Board was flashed and verified successfully.

dreimer1986, 2019-11-20:
OK, and as always you were 100% right @Koenkk ^^ Network is fully back by now after a complete repairing with new pan_id. That soft reset timeout I did disable because it triggered everytime. Even disabled all seems to run fine now.

a-bailey, 2019-11-20:
I'll try changing the Pan ID tomorrow and see If that works, what do you mean when talking about the soft reset timeout?

dreimer1986, 2019-11-20:
I have no clue if your problem is the same as mine. I had no problems similar to your log entries on start/stop. But maybe... it fixes your problems too if you had some devices left on another network.
I meant that the soft reset timeout was triggered every 60 sec and then resetted the cc1352p-2 and now I changed it to 0 to stop that.

Looked like that:
warn  2019-11-20 21:42:11: Soft reset timeout triggered
warn  2019-11-20 21:42:13: Soft resetted ZNP due to timeout

Koenkk, 2019-11-21:
@a-bailey it looks that zigbee2mqtt doesn't stop gracefully, (you should see something like stopping zigbee2mqtt). Please open an issue at the hassio addon repo (https://github.com/danielwelch/hassio-zigbee2mqtt/issues)

a-bailey, 2019-11-22:
@Koenkk I'm not using the hassio addon just plain zigbee2mqtt.
My current log shows the following on start:

`Nov 22 16:42:46  systemd[1]: Started zigbee2mqtt.
Nov 22 16:42:48  npm[953]: > zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
Nov 22 16:42:48  npm[953]: > node index.js
Nov 22 16:42:48  npm[953]:                 Zigbee2mqtt requires node version 10, you are running v8.16.1!
Nov 22 16:43:00  npm[953]: zigbee2mqtt:info  2019-11-22T15:43:00: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-11-22.16-42-59'
Nov 22 16:43:00  npm[953]: zigbee2mqtt:info  2019-11-22T15:43:00: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
Nov 22 16:43:00  npm[953]: zigbee2mqtt:info  2019-11-22T15:43:00: Starting zigbee-herdsman...
Nov 22 16:43:01  npm[953]: zigbee2mqtt:error 2019-11-22T15:43:01: Error while starting zigbee-herdsman
Nov 22 16:43:01  npm[953]: zigbee2mqtt:error 2019-11-22T15:43:01: Failed to start zigbee
Nov 22 16:43:01  npm[953]: zigbee2mqtt:error 2019-11-22T15:43:01: Exiting...
Nov 22 16:43:01  npm[953]: zigbee2mqtt:error 2019-11-22T15:43:01: SyntaxError: Unexpected token
Nov 22 16:43:01  npm[953]:     at JSON.parse (<anonymous>)
Nov 22 16:43:01  npm[953]:     at Function.open (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/database.js:17:35)
Nov 22 16:43:01  npm[953]:     at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:87:48)
Nov 22 16:43:01  npm[953]:     at Generator.next (<anonymous>)
Nov 22 16:43:01  npm[953]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:5:58)
Nov 22 16:43:01  npm[953]:     at <anonymous>
Nov 22 16:43:01  npm[953]:     at process._tickCallback (internal/process/next_tick.js:189:7)
Nov 22 16:43:01  npm[953]: npm ERR! code ELIFECYCLE
Nov 22 16:43:01  npm[953]: npm ERR! errno 1
Nov 22 16:43:01  npm[953]: npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`
Nov 22 16:43:01  npm[953]: npm ERR! Exit status 1
Nov 22 16:43:01  npm[953]: npm ERR!
Nov 22 16:43:01  npm[953]: npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
Nov 22 16:43:01  npm[953]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Nov 22 16:43:01  npm[953]: npm ERR! A complete log of this run can be found in:
Nov 22 16:43:01  npm[953]: npm ERR!     /home/pi/.npm/_logs/2019-11-22T15_43_01_374Z-debug.log
Nov 22 16:43:01  systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Nov 22 16:43:01  systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Nov 22 16:43:01  systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Nov 22 16:43:01  systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Nov 22 16:43:01  systemd[1]: Stopped zigbee2mqtt.`

```
test -w /dev/ttyACM0 && echo success || echo failure
success
```

I had to compile the firmware myself (ofc applying your firmware patch before) when using the hex from the repository I can't even write on the port + the uniflash software fails verifying the .hex

Koenkk, 2019-11-22:
It looks like your `database.db` got corrupted.

a-bailey, 2019-11-23:
Should I just delete the database.db and try again or do I also have to delete other files? Is there a "best practise" way to reset z2mqtt?

Koenkk, 2019-11-23:
Empty the data directory and reflashing coordinator with manual erase is a complete factory reset

a-bailey, 2019-11-23:
I'll try that and report back thanks

a-bailey, 2019-11-23:
After clearing the data folder except confiuration which I reset to original state I get this error now:

`Nov 23 16:30:46  systemd[1]: Started zigbee2mqtt.
Nov 23 16:30:47  npm[2661]: > zigbee2mqtt@1.7.1 start /opt/zigbee2mqtt
Nov 23 16:30:47  npm[2661]: > node index.js
Nov 23 16:30:48  npm[2661]:                 Zigbee2mqtt requires node version 10, you are running v11.15.0!
Nov 23 16:30:50  npm[2661]: zigbee2mqtt:info  2019-11-23T15:30:50: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-11-23.16-30-49'
Nov 23 16:30:50  npm[2661]: zigbee2mqtt:info  2019-11-23T15:30:50: Starting zigbee2mqtt version 1.7.1 (commit #b459c35)
Nov 23 16:30:50  npm[2661]: zigbee2mqtt:info  2019-11-23T15:30:50: Starting zigbee-herdsman...
Nov 23 16:30:50  npm[2661]: zigbee2mqtt:error 2019-11-23T15:30:50: Error while starting zigbee-herdsman
Nov 23 16:30:50  npm[2661]: zigbee2mqtt:error 2019-11-23T15:30:50: Failed to start zigbee
Nov 23 16:30:50  npm[2661]: zigbee2mqtt:error 2019-11-23T15:30:50: Exiting...
Nov 23 16:30:50  npm[2661]: zigbee2mqtt:error 2019-11-23T15:30:50: Error: The module '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node'
Nov 23 16:30:50  npm[2661]: was compiled against a different Node.js version using
Nov 23 16:30:50  npm[2661]: NODE_MODULE_VERSION 57. This version of Node.js requires
Nov 23 16:30:50  npm[2661]: NODE_MODULE_VERSION 67. Please try re-compiling or re-installing
Nov 23 16:30:50  npm[2661]: the module (for instance, using `npm rebuild` or `npm install`).
Nov 23 16:30:50  npm[2661]:     at Object.Module._extensions..node (internal/modules/cjs/loader.js:857:18)
Nov 23 16:30:50  npm[2661]:     at Module.load (internal/modules/cjs/loader.js:685:32)
Nov 23 16:30:50  npm[2661]:     at Function.Module._load (internal/modules/cjs/loader.js:620:12)
Nov 23 16:30:50  npm[2661]:     at Module.require (internal/modules/cjs/loader.js:723:19)
Nov 23 16:30:50  npm[2661]:     at require (internal/modules/cjs/helpers.js:14:16)
Nov 23 16:30:50  npm[2661]:     at bindings (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:112:48)
Nov 23 16:30:50  npm[2661]:     at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:2:36)
Nov 23 16:30:50  npm[2661]:     at Module._compile (internal/modules/cjs/loader.js:816:30)
Nov 23 16:30:50  npm[2661]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:827:10)
Nov 23 16:30:50  npm[2661]:     at Module.load (internal/modules/cjs/loader.js:685:32)
Nov 23 16:30:50  npm[2661]: npm ERR! code ELIFECYCLE
Nov 23 16:30:50  npm[2661]: npm ERR! errno 1
Nov 23 16:30:50  npm[2661]: npm ERR! zigbee2mqtt@1.7.1 start: `node index.js`
Nov 23 16:30:50  npm[2661]: npm ERR! Exit status 1
Nov 23 16:30:50  npm[2661]: npm ERR!
Nov 23 16:30:50  npm[2661]: npm ERR! Failed at the zigbee2mqtt@1.7.1 start script.
Nov 23 16:30:50  npm[2661]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Nov 23 16:30:50  npm[2661]: npm ERR! A complete log of this run can be found in:
Nov 23 16:30:50  npm[2661]: npm ERR!     /home/pi/.npm/_logs/2019-11-23T15_30_50_417Z-debug.log
Nov 23 16:30:50  systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Nov 23 16:30:50  systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Nov 23 16:30:50  systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Nov 23 16:30:50  systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Nov 23 16:30:50  systemd[1]: Stopped zigbee2mqtt.`

The system seems to complain about the nodejs version on this pi could that be a source of the problem?

Koenkk, 2019-11-25:
Can you try a `rm -rf node_modules && npm install` in the zigbee2mqtt folder?

a-bailey, 2019-11-25:
Thanks for the help, clearing my node_modules folder gave me a new error "AREQ - ZDO - stateChangeInd after 60000ms" etc. which the user before me had fixed it by changing the pan_id. This is my development pie does this happen because I already have a different pi running zigbee2mqtt?  Haven't paired any devices yet but the log seems to be fine :) 

`Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}}'
Currently 0 devices are joined:
Zigbee: disabling joining new devices.
Connecting to MQTT server at mqtt://localhost:1883
Connected to MQTT server
MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.7.1","commit":"b459c35","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}},"log_level":"info","permit_join":false}'`

closed 2020-08-18 by github-actions[bot]