Koenkk/Z-Stack-firmware #122: CC2531 stopped working for no apparent reason

PedroLamas, 2019-11-02:
I have a Nano PI Neo 2 with a CC2531 attached running zigbee2mqtt via docker for a few months now without any issues at all, but today, for no apparent reason at all, it stopped working!

I think I flashed CC2531_DEFAULT_20190608 in June or so, and it's been on ever since.

If I disconnect it from the USB port and reattach it, I can see it appearing in `/dev/ttyACM0` but after a few seconds, it disappears...

I haven't tried to reflash it yet but will do first thing tomorrow morning, but I wanted to check if anyone else had ever experienced this.

Here's the results of a few commands showing the connection & disconnection of the device
```
pedro@nanopineo2:~/docker$ dmesg -T -w
[Sat Nov  2 00:07:34 2019] usb 8-1: new full-speed USB device number 76 using ohci-platform
[Sat Nov  2 00:07:34 2019] usb 8-1: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[Sat Nov  2 00:07:34 2019] usb 8-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sat Nov  2 00:07:34 2019] usb 8-1: Product: TI CC2531 USB CDC
[Sat Nov  2 00:07:34 2019] usb 8-1: Manufacturer: Texas Instruments
[Sat Nov  2 00:07:34 2019] usb 8-1: SerialNumber: __0X00124B0019366C03
[Sat Nov  2 00:07:34 2019] cdc_acm 8-1:1.0: ttyACM0: USB ACM device
[Sat Nov  2 00:08:35 2019] usb 8-1: USB disconnect, device number 76
[Sat Nov  2 00:08:35 2019] usb 8-1: new full-speed USB device number 77 using ohci-platform
[Sat Nov  2 00:08:51 2019] usb 8-1: device descriptor read/64, error -110
[Sat Nov  2 00:09:07 2019] usb 8-1: device descriptor read/64, error -110
[Sat Nov  2 00:09:07 2019] usb 8-1: new full-speed USB device number 78 using ohci-platform
[Sat Nov  2 00:09:11 2019] hrtimer: interrupt took 1050329 ns
[Sat Nov  2 00:09:23 2019] usb 8-1: device descriptor read/64, error -110
[Sat Nov  2 00:09:38 2019] usb 8-1: device descriptor read/64, error -62
[Sat Nov  2 00:09:38 2019] usb usb8-port1: attempt power cycle
[Sat Nov  2 00:09:39 2019] usb 8-1: new full-speed USB device number 79 using ohci-platform
[Sat Nov  2 00:09:49 2019] usb 8-1: device not accepting address 79, error -110
[Sat Nov  2 00:09:49 2019] usb 8-1: new full-speed USB device number 80 using ohci-platform
[Sat Nov  2 00:10:00 2019] usb 8-1: device not accepting address 80, error -110
[Sat Nov  2 00:10:00 2019] usb usb8-port1: unable to enumerate USB device
[Sat Nov  2 00:10:39 2019] usb 8-1: new full-speed USB device number 81 using ohci-platform
[Sat Nov  2 00:10:55 2019] usb 8-1: device descriptor read/64, error -110
```

```
pedro@nanopineo2:~/docker$ udevadm monitor 
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[1921.720218] add      /devices/platform/soc/1c1d400.usb/usb8/8-1 (usb)
KERNEL[1921.721493] add      /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0 (usb)
KERNEL[1921.723613] add      /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0/tty/ttyACM0 (tty)
KERNEL[1921.723773] bind     /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0 (usb)
KERNEL[1921.723890] add      /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.1 (usb)
KERNEL[1921.723993] bind     /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.1 (usb)
KERNEL[1921.724296] bind     /devices/platform/soc/1c1d400.usb/usb8/8-1 (usb)
UDEV  [1921.893477] add      /devices/platform/soc/1c1d400.usb/usb8/8-1 (usb)
UDEV  [1921.920929] add      /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.1 (usb)
UDEV  [1921.921140] add      /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0 (usb)
UDEV  [1921.924288] bind     /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.1 (usb)
UDEV  [1921.928622] add      /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0/tty/ttyACM0 (tty)
UDEV  [1921.931598] bind     /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0 (usb)
UDEV  [1921.936481] bind     /devices/platform/soc/1c1d400.usb/usb8/8-1 (usb)
KERNEL[1982.579963] remove   /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0/tty/ttyACM0 (tty)
KERNEL[1982.580489] unbind   /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.1 (usb)
KERNEL[1982.580588] unbind   /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0 (usb)
KERNEL[1982.580689] remove   /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0 (usb)
KERNEL[1982.580799] remove   /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.1 (usb)
KERNEL[1982.584237] unbind   /devices/platform/soc/1c1d400.usb/usb8/8-1 (usb)
KERNEL[1982.584442] remove   /devices/platform/soc/1c1d400.usb/usb8/8-1 (usb)
UDEV  [1982.588914] remove   /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0/tty/ttyACM0 (tty)
UDEV  [1982.592601] unbind   /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0 (usb)
UDEV  [1982.595128] remove   /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0 (usb)
UDEV  [1982.598246] unbind   /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.1 (usb)
UDEV  [1982.600381] remove   /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.1 (usb)
UDEV  [1982.610956] unbind   /devices/platform/soc/1c1d400.usb/usb8/8-1 (usb)
UDEV  [1982.613523] remove   /devices/platform/soc/1c1d400.usb/usb8/8-1 (usb)
```

```
pedro@nanopineo2:~/docker$ udevadm info /dev/ttyACM0
P: /devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0/tty/ttyACM0
N: ttyACM0
L: 0
S: serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0019366C03-if00
S: serial/by-path/platform-1c1d400.usb-usb-0:1:1.0
E: DEVPATH=/devices/platform/soc/1c1d400.usb/usb8/8-1/8-1:1.0/tty/ttyACM0
E: DEVNAME=/dev/ttyACM0
E: MAJOR=166
E: MINOR=0
E: SUBSYSTEM=tty
E: USEC_INITIALIZED=2423131859
E: ID_VENDOR=Texas_Instruments
E: ID_VENDOR_ENC=Texas\x20Instruments
E: ID_VENDOR_ID=0451
E: ID_MODEL=TI_CC2531_USB_CDC
E: ID_MODEL_ENC=TI\x20CC2531\x20USB\x20CDC
E: ID_MODEL_ID=16a8
E: ID_REVISION=0009
E: ID_SERIAL=Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0019366C03
E: ID_SERIAL_SHORT=__0X00124B0019366C03
E: ID_TYPE=generic
E: ID_BUS=usb
E: ID_USB_INTERFACES=:020201:0a0000:
E: ID_USB_INTERFACE_NUM=00
E: ID_USB_DRIVER=cdc_acm
E: ID_USB_CLASS_FROM_DATABASE=Communications
E: ID_VENDOR_FROM_DATABASE=Texas Instruments, Inc.
E: ID_PATH=platform-1c1d400.usb-usb-0:1:1.0
E: ID_PATH_TAG=platform-1c1d400_usb-usb-0_1_1_0
E: DEVLINKS=/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0019366C03-if00 /dev/serial/by-path/platform-1c1d400.usb-usb-0:1:1.0
E: TAGS=:systemd:
```

PedroLamas, 2019-11-02:
Didn't want to wait till morning and went ahead and reflashed it, and... it's working again!

It does seem that the chip firmware got corrupted... has anyone ever experienced such an issue?

I'm leaving it open for comments and will update if the situation changes, but feel free to close in the meantime!

Koenkk, 2019-11-02:
If you are having stability issues I would recommend the source routing firmware instead of the default one (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

PedroLamas, 2019-11-02:
Thank you @Koenkk, I will keep monitoring the situation and if it happens again will do just that!

One thing I did notice is that the Salus SP600 is quite "chatty" as it sends power measurements (more than 1 per second), I wonder if it has anything to do with [these](https://github.com/Koenkk/zigbee-herdsman-converters/blob/6306a93079355f209508251fa6c28752fdcbdeda/devices.js#L4148) numbers.

I'm also considering moving from shepherd to herdsman (so from stable docker image to dev one) to see if there is any difference.

PedroLamas, 2019-11-02:
I've changed those values in my local devices.js to match the same used in the other smart plugs:

```javascript
configure: (ieeeAddr, shepherd, coordinator, callback) => {
    const device = shepherd.find(ieeeAddr, 9);
    const onOff = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
    const actions = [
        (cb) => device.foundation('genOnOff', 'configReport', [onOff], foundationCfg, cb),
        (cb) => device.bind('genOnOff', coordinator, cb),
        (cb) => device.report('seMetering', 'instantaneousDemand', 10, 60, 1, cb),
    ];
    execute(device, actions, callback);
},
```

This made it less "chatty" (only one update per 10s, which I belive is good enough).

(On a side note, can't wait to start using the herdsman typescript based stuff to have proper typing and thus understand what all these "magic numbers" mean!)

allofmex, 2019-11-02:
> If I disconnect it from the USB port and reattach it, I can see it appearing in /dev/ttyACM0 but after a few seconds, it disappears... I wanted to check if anyone else had ever experienced this.

I had this twice in the last week's too. Only reflash worked for me. Did not really found a reason yet.

PedroLamas, 2019-11-02:
Did anything change in your network? Did you add any new device to it?

I have a feeling my problem was the SP600 being so chatty, so let's see if the new values help!

allofmex, 2019-11-02:
Yes, I added new (6 hue bulbs) short before it crashed the first time.
Now ~30 devices. But the last two weeks were stable again. 
I assume it was more related to a sudden broker (docker container) crash. But it's just a guess.

Koenkk, 2019-11-03:
@PedroLamas the reporting configuration is different then in 1.6 so this should probably fix the issue.

If you are still experiencing crashes with the source routing firmware (and the `report` and `availability` feature disabled) you just got to the limits of the CC2531, a solution would be to switch to the much more powerful CC2652R or CC1352-P (https://github.com/koenkk/zigbee2mqtt/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopen+cc2652r)

closed 2019-11-03 by Koenkk