Koenkk/Z-Stack-firmware #123: Error: AF data request fails, status code: 205. No network route.

jendib, 2019-11-02:
I get this "Error: AF data request fails, status code: 205. No network route." error randomly on random lights (Philips Hue and Innr) and the switching fails when this happen.
I tried rebooting the CC2531 dongle, restarting zigbee2mqtt, repairing the lights.

Any idea? Thanks.

Koenkk, 2019-11-03:
When experiencing 205 errors it could help to use the source routing firmware instead of the default one.

jendib, 2019-11-03:
The dongle was behind my home server, below the Wifi router, in a bunch of cables. I pulled it out of this with an USB extender and it seems much better.
I will try the source routing firmware if I still experience issues.
Thanks! :)

closed 2019-11-03 by jendib