Koenkk/Z-Stack-firmware #125: SONOFF BASICZBR3 - A new pre-built router?

marrold, 2019-11-06:
Sonoff have released the BASICZBR3, a Zigbee device based on the CC2530 - https://www.youtube.com/watch?v=yFwJAQFndEo

Assuming the pins are exposed to program it, this could make a nice "pre-built" router

ptvoinfo, 2019-11-07:
Yes, but it would be great to have a chip with an amplifier there.

marrold, 2019-11-15:
Seems like the Zigbee range / performance is actually fairly disappointing.

drogfild, 2019-11-17:
Off topic: is it even worth buying for a Zigbee switch?

marrold, 2019-11-17:
> Off topic: is it even worth buying for a Zigbee switch?

It's cheap and it works. RF performance is probably good enough for an End Device assuming you have reasonable network coverage. It's just not quite good enough to expand signal coverage as a router, but in fairness that's not the devices intention.

closed 2019-12-11 by marrold

Gamester17, 2020-05-25:
FYI, kirovilya has released Z-Stack 3.0 firmware/source with Zigbee 3.0 support for Sonoff BASICZBR3.

Please see related request/discussion -> https://github.com/Koenkk/Z-Stack-firmware/issues/176