Koenkk/Z-Stack-firmware #126: Migrated from very old firmware (20180815 to latest 20190619 SOURCE ROUTING)

quarcko, 2019-11-14:
Hi,

So as the title says i migrated my setup from 20180815 to 20190619 SOURCE ROUTING.
Before the migration and with latest 1.7.1 zigbee2mqtt my system was rock solid
with immediate sensor responds (my all devices are Aqara ones, including some 10 routers
and about 20 end devices).

Migration i did:
1) new stick with new firware.
stopped zigbee2mqtt, replaced stick, changed COM port number and started.
After doing so, tried to toggle some switches, open some doors or press some wireless buttons,
some of them seems to be working, so no pairing is needed as advertised.

But overall stability of my sensors is awful compared to my old stick with old firmware.
Some door sensors respond in like 3-5 seconds, and after closing a door
they keep "open" state for more than 1 hour before reporting correct state.
Also there is frequent "LQI scan failed" for some routers, then after some time
scan succeds, then again fails. It happens to different routers at different times.
So my MAP is sometimes complete sometimes not.

Also - i had to manually edit the "database.db" because my coordinator MAC adress was left from old stick, so edited this file by hand and replaced all occurencies of old MAC with new MAC.

After doing so - no improvements... system is anyway unstable with huge delays from sensors.

So good for me that i have my old stick untouched, thinking of putting it back.
Old firmware seems to be way better than latest one...

Is this supposed to be so, maybe i could provide you some logs or something?

quarcko, 2019-11-14:
This is for example image of Aqara wall outled with measured power consumption.
It runs my heating on it, and you can see with green square data from old stick
and yellow from new stick... there is A LOT of missing data, long periods "stuck"
(marked with red) when data was not incoming at all....

What i'm doing wrong? maybe there are needed some special steps
when migrating from physically one stick to another?

[image]

quarcko, 2019-11-14:
And what is this:
zigbee2mqtt:debug 2019-11-14T14:28:55: Skipping re-transmitted Xiaomi message

Maybe it is my missing data?

Koenkk, 2019-11-15:
> Also there is frequent "LQI scan failed" for some routers, then after some time
scan succeds, then again fails. 

How often do you do network scans?

quarcko, 2019-11-19:
I do them once every 30 min.
But on old stick there is no problem with them, that's what's strange.

Now i returned old stick back and my network is again rock solid,
Also there is no more "Skipping re-transmitted Xiaomi message"
So for now i will be on old firmware as it looks to perform better.

antonakv, 2020-04-02:
I'm curious if it worth to migrate as well @Koenkk #3154

quarcko, 2020-08-10:
Hi @Koenkk ,

Recently we talked about Tradfri remote, and i mentioned, that new firmware performs poorly for me,
So just pinging this bug report, as it is still unsolved, and now it seems i really need this upgrade.

So how can we dig further to investigate the issue?

Koenkk, 2020-08-11:
@quarcko how many devices do you have in your network? Have you also tried the default firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default

quarcko, 2020-08-11:
I have 35 end devices (all of them Xiaomi stuff)
and 12 routers (11 Xiaomi, and 1 Tradfri bulb)
This report is actually quite old, and when i first tried new firmware my network was a bit smaller but still misbehaving.

Yes i have tried also default firmware with same results.
I currently have 3 separate sticks with 3 different firmwares loaded into each of them, for testing.

So, any advanced debug to check for?

Koenkk, 2020-08-11:
One thing you can also try is to repair all devices (I know this is not nice but maybe solves the problem). I would recommend using the source routing firmware.

quarcko, 2020-08-11:
I might try some experiment like this:
I will find some device that is really slow to respond with source routing firmware
and will repair only this particular device and see if there are any improvements.. ?

Koenkk, 2020-08-11:
I would recommend starting from scratch, it's a really strange problem which I haven't seen reports of yet.

quarcko, 2020-08-11:
like... reinstalling whole addon with no config and *.db file and anything?

Koenkk, 2020-08-12:
Yes, other things you can try is to just repair the problematic device (remove it first through https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove) or sniff to see where things go wrong: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

quarcko, 2020-08-15:
So i think this can be closed... steps that helped:

1) Soldered SMA connector and added WIFI antenna to stick
2) After this started z2m, tested for a while, seemed a bit better, but still lagging was felt
     Also there were 40% not responding routers with failed LQI scans.
3) Powered down all my routers while coordinator was running (for some 10mins)
4) Started all routers at once. (really just powered down and up whole house)
5) Repaired 2 end devices that were not sending data at all.

Now everything seems to be working quite good. There is occasional micro-lag here and there
but seems almost as good as old stick were :)) woohoo :)

One thing that bothers me - with SOURCE routing firmware there is like very limited number of direct
children, right? But with new setup coordinator somehow has chosen a direct connection to 2 end devices
instead of going via router way. Hope that he can shake that off in case there is need.
Old setup did not have any end devices directly to coordinator. only via router connections.

In a week my network will expand with 6 more routers, hope that all will go well :)

Thanks again @Koenkk for a great software!

Koenkk, 2020-08-17:
> One thing that bothers me - with SOURCE routing firmware there is like very limited number of direct
children, right? 

Yes only 5 indeed. 

Good that it works now!

closed 2020-08-17 by Koenkk