Koenkk/Z-Stack-firmware #127: Using CC2530+CC2591 as a router - which pins are used for UART ?

marrold, 2019-11-17:
I'm considering using one of these [CC2530+CC2591](https://www.ebay.co.uk/itm/Anti-interference-TI-CC2530-CC2591-UART-ZigBee-Module-Wireless-w-2-4G-Antenna/163827817866) devices as a router.

I can find the firmware in this repo, but it doesn't document which of the CC2530s pins are used for the UART interface. I did find [this comment](https://github.com/Koenkk/zigbee2mqtt/issues/52#issuecomment-390658072) - is it still relevant? 

Thanks

ptvoinfo, 2019-11-18:
@marrold The mentioned router firmware does not support the UART interface.

marrold, 2019-11-18:
@ptvoinfo Yes my mistake, I meant a coordinator. 

Thanks

ptvoinfo, 2019-11-18:
Coordinator: P02, P03, GND

closed 2019-11-18 by marrold