Koenkk/Z-Stack-firmware #131: CC2531 - No network route

sztof, 2019-12-10:
Hi,
I am using a SPZB0001 thermostat connected to zigbee network. There are some problems with zigbee range so I bought two Ikea Tradfri smart plugs to make them work as repeaters.
It looks like the range didnt changed from time to time i have "No network route (205)" error from SPZB0001. 
All of the devices are succesfully paired in the network. I have also tried few firmware versions od CC2531, but with no effect.
In the same (small) room with SPZB0001 there is also xiaomi temp/humidity sensor and it works all the time without any problems.
Do you have any ideas what may be wrong?

allofmex, 2019-12-11:
What other router/powered devices are in your network?

sztof, 2019-12-11:
There are only two ikea tradfri routers,
Here is list of all devices in the network:

SPZB0001 - Eurotronic Spirit Zigbee wireless heater thermostat (EndDevice)
E1603/E1702 - IKEA TRADFRI control outlet (Router)
E1603/E1702 - IKEA TRADFRI control outlet (Router)
Xiaomi MiJia temperature & humidity sensor (EndDevice)
Xiaomi MiJia temperature & humidity sensor (EndDevice)

One of temperature sensors and Tradfri  sockets is in the same room as SPZB0001.
But range problem (I dont know why) concerns only SPZB0001.

allofmex, 2019-12-11:
> only two ikea tradfri

Ok, this should be fine. I had troubles with OSRAM and others, but not with ikea.

What cc2531 firmware are you using? default or source_routing? (see also https://github.com/ioBroker/ioBroker.zigbee/issues/418)

Firmware of your Eurotronic?

sztof, 2019-12-11:
Right now I am using cc2531 with 3.0 firmware CC2531_20190425.zip.
But the same problem persist on other versions also.

`Coordinator firmware version: '{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20190425}}'
`
Unfortunetly, I dont know how to check Eurotronic firmware

allofmex, 2019-12-11:
> from time to time I still have "No network route (205)"

What time span do you talk about? hours/days/weeks?
To be honest, my Eurotronics are not super stable too, from time to time, the one or the other needs a power reset to rejoin (maybe 1 in a week or two). But I'm on an old firmware, would be good to know yours.

You did not said what system you are running, in case of zigbee2mqtt:
> You can get the firmware info by doing an empty publish to zigbee2mqtt/bridge/config/devices/get

And finally some info regarding 3.0 on cc2531: https://github.com/ioBroker/ioBroker.zigbee/issues/380

sztof, 2019-12-17:
After few restarts and changing ikea tradfri positioning SPZB0001 started to work better, "no network route" error did not appeared anymore. 
Unfortunetly Im am not quite sure what helped here.

closed 2020-08-18 by github-actions[bot]