Koenkk/Z-Stack-firmware #132: Provide Firmware 1.7.1 as new bin file

Ingo1977, 2019-12-11:
Hello,

can you please provide the latest firmware 1.7.1 as compiled file?

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default

For some reason I am not able to compile the source.

Thanks in advance.

Ingo1977, 2019-12-18:
Hello,
I changed the subject. Can someone please compile and share the latest bin file for v1.8.0 with me? I tried but I am not able to compile it. 
Thanks.

Koenkk, 2019-12-19:
You can find the `.bin` file in the zip you linked (e.g. https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2531_DEFAULT_20190608.zip) This can be used with zigbee2mqtt 1.8.

closed 2020-08-18 by github-actions[bot]