Koenkk/Z-Stack-firmware #133: Starting cc1352p-2 after reboot

kvet, 2019-12-13:
Is it a way to boot cc1352p-2 firmware on power on? Now I need to click a button near micro-usb port each time my raspberry pi is rebooted

Koenkk, 2019-12-14:
Unfortunately I cannot reproduce this issue (on a raspberry pi with t cc1352p-2). 

However can you try replacing `/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js` with https://gist.github.com/Koenkk/7452bb8145a549deb9754e912155c4a1

Please provide the herdsman debug log if it doesn't work.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

kvet, 2019-12-19:
@Koenkk It didn't helped me.

Here is what I am getting with enabled debug logging:

```
Using '/app/data' as data directory

> zigbee2mqtt@1.8.0 start /app
> node index.js

winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
winston:create-logger Define prototype method for "error"
winston:create-logger Define prototype method for "warn"
winston:create-logger Define prototype method for "info"
winston:create-logger Define prototype method for "http"
winston:create-logger Define prototype method for "verbose"
winston:create-logger Define prototype method for "debug"
winston:create-logger Define prototype method for "silly"
zigbee2mqtt:info  2019-12-19 18:54:04: Logging to console and directory: '/app/data/log/2019-12-19.18-54-04'
winston:file written true false
zigbee2mqtt:debug 2019-12-19 18:54:04: Removing old log directory '/app/data/log/2019-12-19.18-48-08'
winston:file written true false
zigbee2mqtt:debug 2019-12-19 18:54:04: Loaded state from file /app/data/state.json
winston:file written true false
winston:file logged 97 info  2019-12-19 18:54:04: Logging to console and directory: '/app/data/log/2019-12-19.18-54-04'

winston:file logged 187 debug 2019-12-19 18:54:04: Removing old log directory '/app/data/log/2019-12-19.18-48-08'

winston:file logged 258 debug 2019-12-19 18:54:04: Loaded state from file /app/data/state.json

winston:file ENOENT ok /app/data/log/2019-12-19.18-54-04/log.txt
winston:file stat done: log.txt { size: 0 }
winston:file create stream start /app/data/log/2019-12-19.18-54-04/log.txt { flags: 'a' }
winston:file create stream ok /app/data/log/2019-12-19.18-54-04/log.txt
zigbee2mqtt:info  2019-12-19 18:54:05: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)
winston:file written true false
zigbee2mqtt:info  2019-12-19 18:54:05: Starting zigbee-herdsman...
winston:file written true false
zigbee2mqtt:debug 2019-12-19 18:54:05: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[33,117,141,25,0,75,18,0],"channelList":[26],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":null}}'
winston:file written true false
2019-12-19T15:54:05.047Z serialport/bindings loading LinuxBinding
2019-12-19T15:54:05.127Z zigbee-herdsman:adapter No path provided, auto detecting path
2019-12-19T15:54:05.129Z serialport/stream .list
winston:file logged 80 info  2019-12-19 18:54:05: Starting zigbee2mqtt version 1.8.0 (commit #da4d26a)

winston:file logged 135 info  2019-12-19 18:54:05: Starting zigbee-herdsman...

winston:file logged 510 debug 2019-12-19 18:54:05: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[33,117,141,25,0,75,18,0],"channelList":[26],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":null}}'

winston:file file open ok /app/data/log/2019-12-19.18-54-04/log.txt
2019-12-19T15:54:05.240Z zigbee-herdsman:adapter Auto detected path '/dev/ttyACM0' from adapter 'ZStackAdapter'
2019-12-19T15:54:05.242Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[33,117,141,25,0,75,18,0],"channelList":[26]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json"}'
2019-12-19T15:54:05.247Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":true,"autoOpen":false}
2019-12-19T15:54:05.250Z serialport/stream opening path: /dev/ttyACM0
2019-12-19T15:54:05.251Z serialport/binding-abstract open
2019-12-19T15:54:05.253Z serialport/stream _read queueing _read for after open
2019-12-19T15:54:05.254Z serialport/bindings/poller Creating poller
2019-12-19T15:54:05.255Z serialport/stream opened path: /dev/ttyACM0
2019-12-19T15:54:05.255Z serialport/stream _read reading { start: 0, toRead: 65536 }
2019-12-19T15:54:05.257Z serialport/binding-abstract read
2019-12-19T15:54:05.257Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2019-12-19T15:54:05.258Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2019-12-19T15:54:05.258Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2019-12-19T15:54:05.259Z serialport/stream _write 1 bytes of data
[... 5 lines cut ...]
2019-12-19T15:54:05.264Z serialport/stream binding.write write finished
2019-12-19T15:54:05.264Z serialport/bindings/unixRead waiting for readable because of code: EAGAIN
2019-12-19T15:54:05.265Z serialport/bindings/poller Polling for "readable"
2019-12-19T15:54:06.267Z zigbee-herdsman:adapter:zStack:znp:AREQ --> SYS - resetReq - {"type":1}
2019-12-19T15:54:06.270Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,1,65,0,1,65]
[... cut ...]
winston:file written true false
zigbee2mqtt:error 2019-12-19 18:54:36: Exiting...
winston:file written true false
zigbee2mqtt:error 2019-12-19 18:54:36: Error: AREQ - SYS - resetInd after 30000ms
    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:44:24)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
winston:file written true false
```

Koenkk, 2019-12-23:
@fredrikgk  do you have any idea?

fredrikgk, 2019-12-24:
@Koenkk, I know I have asked before, but I do not remember the answer. Are you doing a full ZNP reset as the initial command when starting Herdsman? 

I think the problem is that when the rPi is rebooted the USB power down is not long enough to trigger a brown-out reset on the CC13x2/CC26x2. Thus the Zigbee Stack keeps running and is not in a known state when Herdsman fires up again. 

This is just a guess though, I have not done any debugging myself.

fredrikgk, 2019-12-24:
@kvet , one thing you can try is to remove the big yellow tantalum capacitor on the board. It will not have any noticeable impact on performance, but will make the IC reset itself faster when power is removed.

kvet, 2019-12-24:
@Koenkk @fredrikgk I can confirm that if rPi is down for a longer time (than just reboot), it is working with updated `zStackAdapter.js`

Koenkk, 2019-12-26:
@kvet are you sure that is the fix, can you confirm by testing it again without?

fredrikgk, 2020-01-03:
@Koenkk , what changes did you make to zStackAdapter.js?

@kvet , did you confirm the fix by doing another test with the old version?

kvet, 2020-01-04:
@Koenkk @fredrikgk I've just tested everything without applied fix and it is working well. Not sure that there was something related that changed in 1.8 but now it seems to work ok even doing reboot or unplugging for 30 seconds. Seems like the issue is resolved

dzungpv, 2020-01-21:
> 2019-12-19T15:54:05.250Z serialport/stream opening path: /dev/ttyACM0

@kvet Are you wire CC1352P2 to PI 's UART?

kvet, 2020-01-21:
@dzungpv No. I am using USB on rPI and mUSB on CC1352P2

closed 2020-08-11 by kvet