Koenkk/Z-Stack-firmware #134: Upgrade from CC2531 to CC2530/CC2591 not working

zp-markusp, 2019-12-16:
Hi,

thanks of all for this great project!
I'm running it now for a while on a raspberry zero but would like to extend the range a bit. Therefore I bought this device which from my understanding should work: https://de.aliexpress.com/item/4000136938567.html?

    zigbee2mqtt:error 12/15/2019, 6:58:18 AM Error while starting zigbee-shepherd! (Error: SRSP - SYS - osalNvRead after 6000ms)
    zigbee2mqtt:error 12/15/2019, 6:58:18 AM Press the reset button on the stick (the one closest to the USB) and start again
    zigbee2mqtt:error 12/15/2019, 6:58:18 AM Failed to start
	{"message":"SRSP - SYS - osalNvRead after 6000ms","stack":"Error: SRSP - SYS - osalNvRead after 6000ms\n    at Timeout.object.timer.setTimeout [as _onTimeout] (/app/node_modules/zigbee-herdsman/dist/znp/znp.js:227:24)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
    zigbee2mqtt:error 12/15/2019, 6:58:18 AM Exiting...

Device itself seems to be found (CC2531 was running as ttyACM0)
pi@zigbee2mqtt:/opt/zigbee2mqtt/data $ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 Dec 16 09:17 usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB0

     pi@zigbee2mqtt:/opt/zigbee2mqtt/data $ test -w /dev/ttyUSB0 && echo success || echo failure
    success

     pi@zigbee2mqtt:/opt/zigbee2mqtt/data $ docker exec -it zigbee2mqtt test -w /dev/ttyUSB0 && echo success || echo failure
    success

So write permissions seems to be ok, even inside the container.

Config adapted to ttyUSB0:

    homeassistant: true
    permit_join: true
    mqtt:
      base_topic: zigbee2mqtt
      server: 'mqtt://homesrv:1883'
      user: mqtt
      password: mqtt
      client_id: Zigbee
      include_device_information: true
    serial:
      port: /dev/ttyUSB0
    advanced:
      last_seen: ISO_8601
      log_level: debug

But somehow I don't get the thing up and running.

I used the following firmware: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2530_CC2591_DEFAULT_20190608.zip but also tried different ones.

    DE-MUC1-MAC0025:cc-tool markuss$ sudo ./cc-tool -e -w CC2530ZNP-Prod.hex 
    Programmer: SmartRF04DD
    Target: CC2530
    Erasing flash...
    Completed       
    Writing flash (241 KB)...
    Completed (8.36 s.)

From what I understood from other issues the stick itself doesn't seem to respond accordingly.
Any idea about any further steps to localize the root cause?
Is there any option to verify that the flash process was successful?
Anything I'm missing?

Thanks, Markus

brendanmullan, 2020-01-22:
You're missing this line from your config, which is needed for the CC2530

```
advanced:
  rtscts: false
```

CSauter, 2020-01-31:
> You're missing this line from your config, which is needed for the CC2530
> 
> ```
> advanced:
>   rtscts: false
> ```

Thank you so much.
After Flashing a new CC2530/CC2591 with router-hex I was wondering what I could have missed because the instructions are clear and straight forward. I got no connection to the new router.

After adding your hint and a reboot the router was immediately available.

Thank you again.
Carsten

brendanmullan, 2020-01-31:
@CSauter You're very welcome, glad I could help :-)

ptvoinfo, 2020-02-01:
@CSauter it is strange because this setting is applicable to a coordinator only.

CSauter, 2020-02-03:
> @CSauter it is strange because this setting is applicable to a coordinator only.
Mhm, OK, I'm using a CC2531 as a coordinator since  I have started my zigbee network. So it seems that I missed this setting from scratch?
There were no issues with this missing setting until I tried to add the CC2530 as a router.

closed 2020-08-18 by github-actions[bot]