Koenkk/Z-Stack-firmware #137: Z-Stack_3.0.x brick my CC2531..

Alexofhiswalnut, 2020-01-02:
Hello,
I recently flashed the following firmware: Z-Stack-firmware / coordinator / Z-Stack_3.0.x / bin / CC2531_20190425.zip

But my CC2135 has its green LED which turns off after 60 seconds, even by flashing the previous functional firmware https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/ bin / default / CC2531_DEFAULT_20190608.zip

Do you have a solution to restart my CC2135?

Thank you.

tabnul, 2020-01-04:
I had a comparable problem, see here and how i solved it. might be something to try
https://github.com/Koenkk/Z-Stack-firmware/issues/138

Alexofhiswalnut, 2020-01-05:
Hello, I still have the same problem. Despite the flash of one of the 3 firmware in the following zip file: https: //github.com/Koenkk/Z-Stack-firmware/blob/master/router/CC2531/bin/CC2531_router_2019_02.zip
The original firmware does not work. After 60 seconds the green led always goes out.

it's annoying.

closed 2020-08-18 by github-actions[bot]