Koenkk/Z-Stack-firmware #142: Firmware Patch differences

docbobo, 2020-01-09:
I just noticed some minor difference between the 3.0.x and 3.x.0 stack's patch files. The 3.0.x stack changes afIncomingData in such a way, that messages that cannot find an endpoint will be sent to endpoint 1. In addition, there is some special handling when DstEndPoint is either 10 or 11 - that fragment is missing in the 3.x.0 patch file. 

Was just curious to understand the differences.

Koenkk, 2020-01-10:
This is because it's shouldn't be needed anymore with zigbee-herdsman which came after 3.0.x and before 3.x.0, that's why 3.0.x still has it (and as it is only a small change with no functional impact I don't want to release a new firmware version for it).

docbobo, 2020-01-10:
👍

closed 2020-01-10 by docbobo