Koenkk/Z-Stack-firmware #145: Flashing a Sonoff Basiczbr3

raresserban, 2020-02-11:
Hello,
Just discovered this firmware and I am new to the whole Zigbee DIY ecosystem, only had experience with ESP8266 so far.
I am thinking of buying the Sonoff Basiczbr3, but I have a Philips Hue Hub and I know by default it does not work.
Is it possible to flash it with this firmware, and if so, how?
Thank you and sorry if this is not the right place to ask this.

Gamester17, 2020-02-12:
+1 I am also wondering if "router" Z-Stack-firmware by @Koenkk can be flashed on Sonoff Basic ZBR3?

https://www.itead.cc/sonoff-basiczbr3.html

As I understand Sonoff BASICZBR3 uses a CC2530 or CC2531 Zigbee chip from Texas Instruments and as such probably has an older (1.2.x) version of Z-Stack firmware, but the latest custom Z-Stack firmware by Koenkk also supports running a custom 3.x version of the Z-Stack firmware on CC2530 and CC2531 chips, meaning that hopefully the custom Z-Stack router firmware from Koenkk could probably, in theory, be modified to run the new Sonoff Basic ZBR3?

If possible then these could possibly be great as "routers" to extend the range in a Zigbee mesh-network.

FYI, I also sent this question in as a to request directly to Jerry Shi at Itead/Sonoff who at least replied saying that he would forward the request to the engineers in their research department.

- https://www.reddit.com/r/sonoff/comments/elpueo/sonoff_basiczbr3_could_support_zigbee_30_or/

More information:
- https://notenoughtech.com/home-automation/sonoff-basic-r3-zigbee/
- https://notenoughtech.com/home-automation/sonoff-basic-r3-teardown/

kirovilya, 2020-02-12:
Sonoff BASICZBR3 is always a router, why else flash it?

Gamester17, 2020-02-12:
Sonoff BASICZBR3 currently comes with Z-Stack 1.2 firmware with a "Zigbee Home Automation" profile so it does not support the Zigbee 3.0 profile. Therefore, I want to flash it with Z-Stack 3.x firmware because I want it to be a Zigbee 3.0 profile router instead of just a router for the "Zigbee Home Automation" profile.

Other than making Sonoff BASICZBR3 act as a Zigbee 3.0 router another benefit to non-Zigbee2mqtt users is that it should as well make it compatible with other hubs/gateways/bridges that do not support the "Zigbee Home Automation" profile, such as the Philips Hue Bridge, (I am also pretty sure IKEA Trådfri Gateway do not support Sonoff BASICZBR3 because it uses a "Zigbee Home Automation" profile).

fi-sch, 2020-03-01:
@Gamester17 
Sorry for the off-topic but I've just realised that I don't fully get one thing. Do this firmware: [Koenkk/Z-Stack-firmware/tree/master/router/CC2530_CC2591/bin](https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2530_CC2591/bin) support Z-Stack 3.0? I thought that it's Z-Stack 1.2. Or are you talking about coordinator firmware ([Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin)?

Thanks! And sorry for OT once again.

Hedda, 2020-04-16:
@fi-sch See https://github.com/Koenkk/Z-Stack-firmware/issues/153 Summery; No, there is not yet a Z-Stack 3.0 (3.0.x) based router firmware available for CC2530 and CC2531

kirovilya, 2020-04-16:
I am wrote article about alternative firmware for this Sonoff :)
But in russian https://habr.com/ru/company/iobroker/blog/495926/

Gamester17, 2020-05-20:
Any chance that you can post binary hex file(s) that are ready to be flashed by non-developers too?

Zigbee router firmware hex for Sonoff BASICZBR3 (a.k.a. Sonoff BASIC ZBR3 / Sonoff BASIC ZB R3)?

Would be nice if also could submit to Koenkk repo:
* https://github.com/Koenkk/Z-Stack-firmware/

Gamester17, 2020-05-26:
Moved firmware request discussion to #176

closed 2020-08-18 by github-actions[bot]