Koenkk/Z-Stack-firmware #146: Source routing firmware device limitations

djwlindenaar, 2020-02-14:
I'm considering switching my network over to using a CC2531 with source routing firmware, but I'm concerned with the amount of devices that can be connected. I read that it can support 5 direct children and 40 routes. I have a lot of routers in my network, so that will not be the issue.

My question is what does it mean that it is limted to 40 routes? Does it mean the complete network can only have 40 devices?
Why is that the case?  Still a memory issue of the CC2531? 

Could I re-balance that to support even fewer direct children and support more devices?

Thanks for helping!

Koenkk, 2020-02-14:
> Does it mean the complete network can only have 40 devices?

No, it can store routes to up to 40 devices, having a route improves the efficiency when controlling a device (as it does not have to retrieve the route before controlling), this is only useful for routers.

> Why is that the case? Still a memory issue of the CC2531?

Not really an issue, it's just a limitation of the device itself. If you want to build networks of 40+ devices I would recommend the CC26XR1 or CC1352P-2 (https://www.zigbee2mqtt.io/information/supported_adapters.html#texas-instruments-cc26x2r1)

djwlindenaar, 2020-02-15:
So I have a network of 41 routers and 12 end devices, so I guess that counts as a large network for a CC2531. Maybe time for me to dive into the code to understand what is behind the efficiency remark. I'm not completely clear on what is handled by the coordinator firmware and what is handled in ZigBee2mqtt code. Anywhere you recommend to start?

Currently I'm running it with a DeConz coordinator, but I'm slowly getting fed up with minor stability issues. Although up to now the support has been great, it's simply not open source, so I cannot contribute. Which is becoming a bit frustrating.

Would you say the new CC26XR1 is ready for'production use' (meaning my wife doesn't get pissed because of lights not working)?

imdos, 2020-02-15:
Everything above a cc2531/0 is an improvement. So even the cc2538 will bring a noticeable difference. That one is cheaper than the CC26XR1 or CC1352 but also provides a faster chip. All those chips should benefit from the major improvements in software, which do include stability. I geuss the improvement in hardware provides the most stability. 

I'm unable to judge if this currently is more stable than deconz.

Koenkk, 2020-02-15:
> Would you say the new CC26XR1 is ready for'production use' (meaning my wife doesn't get pissed because of lights not working)?

Yes, definitely, in your case I would really recommend the CC26XR1. The CC2531 is running the complete zigbee stack (so almost everything runs on there), the CC2531 is just too much limited by the hardware/processor on it, especially for larger network.

djwlindenaar, 2020-02-16:
OK. Thanks for the clarifications/advice.

closed 2020-02-16 by djwlindenaar