Koenkk/Z-Stack-firmware #148: CC2538 missing from Z-Stack-firmware coordinator README.md

Gamester17, 2020-02-20:
CC2538 + CC2592 missing from Z-Stack-firmware coordinator README.md or left out on purpose?

Reference #144 as well as the related https://github.com/Koenkk/zigbee2mqtt/issues/1568

Koenkk, 2020-02-20:
Not left out on purpose, it was missed in the original pr, could you make a pr to update it?

Gamester17, 2020-02-21:
PR supported here https://github.com/Koenkk/Z-Stack-firmware/pull/149

@Koenkk @reverieline Please understand that some information about CC2538 support might still be missing as I note below:

Please note that I am not sure if Z-Stack 3.0.x is only supported on a module with on CC2538 + CC2592 or if it is also supported on modules with only CC2538 without an amplifier chip or a module with CC2538 and the CC2591 amplifier chip?

I was not either sure about supported Direct children or the max values for total numbers of Zigbee devices supported by CC2538 in the Z-Stack 3.0.x firmware which has now been added?

In addition, please also note that I am not sure if Zigbee 3.0 coordinator is now actually recommended on a CC2538 module as well, or if it is still only recommended on CC26X2R1 and CC1352P_2 so I did not change that line about such recommendation. CC2538 might be powerful enough to be recommended as well or?

PS: @reverieline Off-topic here but I was wondering if a version of Z-Stack 1.2 could be made for CC2538 too based on your FW?

closed 2020-08-18 by github-actions[bot]