Koenkk/Z-Stack-firmware #150: Failed to flash CC26X2R1 with Z-Stack 3.0

vanlooverenkoen, 2020-02-28:
I just received my CC26x2R1. But I can not figure out how I should flash this board. I tried to use Flash Programmer 2 (windows) and Uniflash (windwows, mac). On all applications my devices can not be detected. When I use lsusb on my mac I can find that a texas instruments device is detected. But never in the software itself. Can someone help me figure out how I can flash this board? Where can I find steps to flash this board from getting it unboxed to instaling z-stack on it.

[image]

pums12, 2020-02-28:
I had the same problem at first as well. It was being detected by Windows but not showing in UniFlash. 

I had to plug it into a USB 2.0 port instead of the USB 3.0 and then it appeared.

vanlooverenkoen, 2020-02-28:
I tried a USB 2.0 and 3.0 port. 

What was your jumper configuration? I can not figure out what jumpers need to be instaled or removed

pums12, 2020-02-28:
I haven't moved any of the jumpers from when I received it. Does it show up in Device Manager?

[image]

vanlooverenkoen, 2020-02-28:
This is what I can see in my device manager. 

The same is detected by lsusb on my mac

[image]

pums12, 2020-02-28:
It only takes seconds for UniFlash to detect it.

This is what shows on mine.

The "Device Firmware Upgrade" device did appear when I had to update the firmware before flashing through UniFlash. At that point, I had to update the drives for the "Device Firmware Upgrade"

[image]

vanlooverenkoen, 2020-02-28:
I do not see the XDS110 ports. 
How did you update the drivers? Because windows indicated the latest/best drivers are installed for Stellaris DeviceFirmware Upgrade

pums12, 2020-02-28:
I can't remember what drives I used. I did install the SDK http://www.ti.com/tool/LAUNCHXL-CC26X2R1 but I think I sorted it before this completely installed.

Take a look at this as it appears the be the same problem but with a different board:
http://e2e.ti.com/support/microcontrollers/other/f/908/t/400294

vanlooverenkoen, 2020-02-28:
Both sollutions did not work

vanlooverenkoen, 2020-02-29:
Alright I figured it out. It seems that my firmware was corrupted. This shoud not pass the productions tests of IT normally. 

- First you need to install the CCS 9 software.
- This will installation will includes the docs, firmware and xdsdfu.exe program. (Uniflash should install this as well but I could not find it anywher)
- The documentation of XDS110SupportReadMe.pdf:
```
Updating the Firmware Using xdsdfu
To program the firmware, follow these steps:
1. Plug the XDS110 debug probe into your computer. Make sure that you only have one
  XDS110 class debug probe plugged in. The xdsdfu program will attempt to flash the first
   XDS110 debug probe it finds.
2. Run the following two commands from directory with xdsdfu:
   xdsdfu -m
   xdsdfu -f firmware.bin -r
You may need to pause after the first command to give the OS time to recognize that the
XDS110 has reconfigured as a different USB device.
Once the second command has completed, the firmware is updated, and the XDS110 should be
ready to use.
Setting the Serial 
```
- Go to the correct path in terminal. For me this was C:\ti\ccs930\ccs\ccs_base\common\uscif\xds110
- Execute `xdsdfu -m`
- Execute `xdsdfu -f firmware.bin -r`
- After that the correct devices are shown in the device manager
[image]

closed 2020-02-29 by vanlooverenkoen

johnfegan, 2020-09-23:
Is there anyway to resolve a bricked device without splashing out 100 quid or so for a XDS110 debug probe? I ordered my LAUNCHXL-CC26X2R1, opened uniflash and was prompted to updated firmware. I did, all seemed well so I followed [zigbee2mqtt instructions](https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html). Again all seemed well but it wasn't recognised by my rpi, plugging it into my pc now it's no longer being detected.

Pascal51882, 2020-09-30:
Same Problem for me.
Device was flashed sucefully and now no Host can detect it all.
Windows detects noting...
The last guide seems not to work because of this.
Does TI return not working devices?

johnfegan, 2020-09-30:
hey @Pascal51882 I finally got mine working, though I'm not 100% what change actually allowed windows to recognise and reflash the device as it took several reboots before this occurred. In the intervening time I had installed:
- simplelink_cc13x2_26x2_sdk_4_20_01_04
- CCS9.1.0.00010_win64
- ti_emupack_setup_8.0.27.9_win_32
I was just wandering forums and trying anything I could find that seemed vaguely relevant. My best guess is one of the above installed compatible drivers for the new firmware. Can't say which order I installed (getting the list from my downloads folder) - I'd given up for a while (and powered on and off a few times in-between as part of natural usage as it's not my main machine). I plugged in again for another round and windows recognised the chip. I reflashed, plugged into rp and it's been running strong 4 days now.

Pascal51882, 2020-09-30:
After installing Simplelink first nothing was changed.
Than I waited and tried again another usb cable and it was detected.
I never had such problems before and flashed CC26x2r 5 times or so.

johnfegan, 2020-09-30:
Bizarre. Glad to hear you're up and running though!

atulp201, 2021-04-30:
Mine, worked out just after disconnecting, waiting for sometime and plugging it again!