Koenkk/Z-Stack-firmware #151: Zigbee 3.0 device to Zigbee 1.2 coordinator

baerni, 2020-03-03:
Hi. Followint this:

> Due to new security requirements in Zigbee 3.0, Zigbee 3.0 coordinators can only support a limited amount of Zigbee 3.0 devices. For Zigbee 1.2 coordinators there is no limit on the max number of Zigbee 3.0 devices that can join.

Can I connect Zigbee 3.0 device to Zigbee 1.2 coordinator? I always thought I could connect the Zigbee 3.0 device to the Zigbee 3.0 coordinator only.

Koenkk, 2020-03-03:
Zigbee is fully backwards compatible so yes this is possible.

closed 2020-03-03 by Koenkk

arteck, 2020-03-04:
@Koenkk  sry but that is not correct. Zigbee 3.0 device don't work with Zigbee 1.2 coordinator
but Zigbee 1.2 device works with Zigbee 3.0 coordinator

Koenkk, 2020-03-05:
@arteck why  not? Many people use e.g. IKEA zigbee 3.0 devices with the Zigbee 1.2 coordinator without problems.

cocooma, 2020-04-02:
I've got the CC1352P2 with the latest firmware Z-Stack_3.x.0 running with zigbee2mqtt 1.12.0. I've migrated all the devices now problem apart from https://www.zigbee2mqtt.io/devices/F-APP-UK-V2.html.

It just does not want to pair with the coordinator. Log is in debug mode but I can not see any attempt to pair at all.

The F-APP-UK-V2 only speaks HA 1.2 It should be backwards compatible.

Do you guys have any idea why?

nonyhaha, 2020-11-27:
@Koenkk sorry to bring this back to life, but it looks like some devices are different from others. As I can connect any Ikea devices to a 1.2 coordinator, I have a friend who bought a new Immax 07047L, zigbee3.0 device, which will not connect to a 1.2 coordinator. Nothing shows in the zigbee2mqtt log. So I am looking for solutions to this, as I think we will have to flash his coordinator to zwave 3,0. What do you think?

Koenkk, 2020-11-27:
@nonyhaha unlikely to be caused by the zigbee version, check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair