Koenkk/Z-Stack-firmware #152: Which firmware/pins to use for a CC2531 used directly as serial, ignoring the provided USB interface?

dewi-ny-je, 2020-03-05:
I have a CC2531 connected to my server. I found  that, even using the provided external antenna, the reach is not enough and I would like to relocate the device.

I have another spot very suitable, but without any server nearby. I thought about using the es8266 as described in the website in association with CC2530.

Can I use the same procedure with CC2531? The debug interface does not provide serial in/out, but there is a row of unpopulated contacts, the first two (p0.2, p0.3) apparently the same used in CC2530 to provide the serial tx/rx.
 
I can solder pins there and check, but is the firmware for CC2530 and CC2531 identical? Or should I reflash it? I'm currently using zstack 1.2 coordinator, default.

If you provide me with sufficient details so that I can test it, I will report the results and edit the wiki page accordingly with all the steps needed.

ptvoinfo, 2020-03-06:
No, you cannot use these pins. The firmware for CC2531 does not include a UART part. The firmware for CC2530 will not work on CC2531.

rudo-thomas, 2020-04-19:
@ptvoinfo your statement contradicts my hands-on experience.

I flashed [CC2530_DEFAULT_20190608](https://github.com/Koenkk/Z-Stack-firmware/tree/4ab0d66e28b10dba75bcc8f1f5a7d7af4fbabf6c/coordinator/Z-Stack_Home_1.2/bin/default) coordinator firmware onto a CC2531 and was able to communicate with via serial it over the P0.2/P0.3 pins. I paired and talked to an end device, which worked too. (Tested with [Tasmota](https://tasmota.github.io/docs/Zigbee/) 8.2.0 on an [Ebyte CC2531 dongle](http://www.ebyte.com/en/product-view-news.aspx?id=523). I only tried powering it through the Vcc/GND pins, not via USB.)

My answer to @dewi-ny-je is: Yes, flash the CC2530 firmware to CC2531, then wire it up as a CC2530, ignoring USB completely.

The fact that CC2530 (coordinator) firmware works on CC2531 doesn't feel like a coincidence, though I haven't read through the source code or datasheets.
Either way, should this fact be mentioned e.g. in the [coordinator README](https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator)?

closed 2020-08-18 by github-actions[bot]

dewi-ny-je, 2020-11-02:
@rudo-thomas can you update the wiki/website pages, in particular https://www.zigbee2mqtt.io/information/supported_adapters.html and https://www.zigbee2mqtt.io/information/connecting_cc2530.html to mention that the CC2531 works too?

Honestly I tried to understand where the source for the website is stored, but I couldn't find it.

rudo-thomas, 2020-11-10:
@dewi-ny-je Sources for those docs are at https://github.com/Koenkk/zigbee2mqtt.io/tree/master/docs/information