Koenkk/Z-Stack-firmware #153: [REQUEST] Z-Stack 3.x router firmware for CC2530 and CC2531?

Hedda, 2020-03-06:
Any chance that you could release Z-Stack 3.0.x router firmware hex files for CC2530 and CC2531?

https://github.com/Koenkk/Z-Stack-firmware/tree/master/router

The reason is that that would like to use some CC2530 and/or CC2531 as Zigbee 3.0 routers.

Is it even possible to use CC2530 or CC2531 hardware as Zigbee 3.0 routers with other firmware?

Hedda, 2020-03-06:
Maybe this request should go to ptvo @ptvoinfo who looks to be credited for existing router firmware?

ptvoinfo, 2020-04-09:
@Hedda There are no advantages of ZStack3 in home environments. Therefore I didn't plan to create a router firmware based on ZStack3.

Hedda, 2020-04-09:
I know that there were no real advantages before today as practically all devices sold for home use used either the ZHA 1.2 or ZLL 1.0 profiles and most devices that used the ZLL profile are also backwards-compliant with the ZHA 1.2 profile, but now more and more Zigbee devices are coming to market that is fully compliant with Zigbee 3.0 certified products and also newer Zigbee coordinator hardware/chips that are fully Zigbee 3.0 compatible.

https://zigbeealliance.org/product_type/certified_product/

Zigbee Alliance also only certify devices that are fully Zigbee 3.0 compliant.

dzungpv, 2020-04-13:
It not possible, TI not update SDK for CC2530/2531/2538 devices to 3.x

Hedda, 2020-04-14:
@dzungpv it is too possible. You are right that TI will not officially release Z-Stack 3.x SDK for older chips like CC2530/2531/2538 but independent developers have already backported Z-Stack 3.x firmware to older chips like CC2530/2531/2538 as you can see here https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

Again it is true that Koenkk is not recommending running CC2530/2531/2538 hardware as a Zigbee 3.0 coordinator in an production environment because of performance issues, but using Z-Stack 3.x firmware as a switch and router firmware on CC2530/2531/2538 hardware might not have the same performance issues are running Z-Stack 3.x coordinator firmware.

dzungpv, 2020-04-14:
> @dzungpv it is too possible. You are right that TI will not officially release Z-Stack 3.x SDK for older chips like CC2530/2531/2538 but independent developers have already inofficially backported Z-Stack 3.x firmware to older chips like CC2530/2531/2538 as you can see here:
> 
> https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator
> 
> Again it is true that @Koenkk is not recommending running CC2530/2531/2538 hardware as a Zigbee 3.0 coordinator in a production environment because of performance issues, but using Z-Stack 3.x firmware as a switch and router firmware on CC2530/2531/2538 hardware might not have the same performance issues are running Z-Stack 3.x coordinator firmware.

It is 3.0.x not 3.x. No dev could backport it, because it is close sourced both in firmware and asic level. You can read  Readme in that respo. Simpler solution, just bought cc1352/cc2652 to use 3.x

Hedda, 2020-04-14:
OK my mistake typing 3.x when I meant 3.0.x to get Zigbee 3.0 router support on CC2530/2531/2538

fi-sch, 2020-04-16:
I would be so glad if @Koenkk would make a router firmware for cheaper cc modules. I personally use a few of cc2530+cc2591 so it would be great to be able to use Z-Stack 3.0 as I add more and more devices to my network.

Gamester17, 2020-05-25:
@kirovilya any chance you are willing to make a Z-Stack 3.0 router firmware for CC2530 and CC2531?

* https://github.com/Koenkk/Z-Stack-firmware/tree/master/router

Just considering that you seem to be the only one who has made Z-Stack 3.0 run on CC253x or?

* https://github.com/diyruz/diyruz_rt

* https://habr.com/ru/company/iobroker/blog/495926/

closed 2020-08-18 by github-actions[bot]

Hedda, 2021-02-16:
Please unstale as I still think quite a few people would like to try Z-Stack 3 router firmware on CC2530 and CC2531 USB dongles.