Koenkk/Z-Stack-firmware #156: CC2652RB does not start with provided image

didiht, 2020-03-20:
Thirst thank you very much for providing the image for the CC2652RB board too.

My board is running well with the self patched and compiled image with the patch from 20200306.

After flashing with the provided image CC2652RB_20200312.hex I always get this error (full log: https://pastebin.com/VZLducvD):

```
zigbee2mqtt:error 2020-03-20 10:07:16: Error: SREQ '--> SYS - osalNvWrite - {"id":3,"len":1,"offset":0,"value":{"type":"Buffer","data":[2]}}' failed with status '10' (expected '0')  
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)  
    at Generator.next (<anonymous>)  
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
```

I tried flashing two times with same result. At the moment I am back to my image.

Koenkk, 2020-03-21:
Thanks for investigating this! It indeed wasn't working and sill needed to find out why :smile:

Fixed firmware is available now.

closed 2020-03-21 by Koenkk

hartmood, 2020-04-14:
@Koenkk I have the same issue with the CC1352P-2

`zigbee2mqtt:error 2020-04-14 13:52:28: Error: SREQ '--> SYS - osalNvWrite - {"id":1,"offset":0,"osal":true,"product":-1,"value":[18,234,165,28,0,75,18,0],"len":8}' failed with status '2' (expected '0')
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:246:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)`

Is there anything I can do myself?

Koenkk, 2020-04-14:
@hartmood make sure to check step 5 from https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html

hartmood, 2020-04-14:
Thanks. But the controller worked a couple of weeks and stopped working the last two days - so why do I have to re-flash it now?

hartmood, 2020-04-14:
I have re-flashed the controller  carefully  step by step. And it works now. I most likely have flashed it previously without steps 4 and 5. For whatever reason it worked before I installed some updates. Many thanks.