Koenkk/Z-Stack-firmware #162: Poor range for CC2530+CC2592PA

KianYeo, 2020-04-18:
I bought a few of these:
https://www.aliexpress.com/item/32563769202.html

And tried to load in this coordinator firmware: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/default/CC2530_CC2592_DEFAULT_20190608.zip

I am using this module with a raspberrypi zero following the instructions here:
https://www.zigbee2mqtt.io/information/connecting_cc2530.html

Everything seems to sort of work except for one major problem. The range is so bad. It only works 2m away from my Xiaomi sensors (I have 2 separate button sensors). Anything more than 2 meters it doesn't receive properly. If I move about 5m away, i will receive multiple random messages of press, double press, long press etc, even though I press and release the button once only.

I have 2 more of these zigbee modules and they both have the same problem. I also tried to use the external SMA antenna by soldering the resistor to connect to the SMA connector and attached an SMA antenna. Its also giving me about 2m range.

Could it be that I am using the wrong coordinator firmware?

ptvoinfo, 2020-04-18:
@KianYeo Please, check the layout of this board. Some Chinese manufacturers use non-standard pins to control CC2592. It works with a proprietary firmware only.

KianYeo, 2020-04-18:
I don't have the schematic of this board, but I guess I can do a visual inspection and use a multimeter.
What should I be looking out for? Do you mean to pin connections between the CC2350 and the CC2592? For this firmware by Koenkk, what pins is it using?

ptvoinfo, 2020-04-18:
Check pins 5,6,7 of CC2592. 
http://www.ti.com/lit/ds/symlink/cc2592.pdf

KianYeo, 2020-04-18:
I found it. The connections on the module that I have are:

PA_EN - P1.3
LNA_EN - P1.2
HGM - P0.0

I presume the coordinator firmware is compiled for another set of pins. Is there anyway I can change the source codes and recompile? Cos I only see the .bin and .hex files here.

ptvoinfo, 2020-04-18:
Koenkk provides some info and patch files:
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

KianYeo, 2020-04-18:
I am looking at the COMPLIE.md file for instructions.
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/COMPILE.md

I am stuck. Have 2 issues.
1. Which file should I open in IAR to change the pin numbers?
2. I followed all the steps exactly from 1 to 8 and tried to rebuild the project (without changing anything files) but I have 2 errors. 

Fatal Error[Pe1696]:cannot open source file "sb_shared.h" (in MT_SYS.c line 76)
Fatal Error[Pe1696]:cannot open source file "sb_shared.h" (in OnBoard.c line 61)

KianYeo, 2020-04-19:
Going to share how I managed to change the pin numbers.  Hopefully it will help others with the same problem.

Basically you need to follow the same instructions in COMPILE.md, then 3 files need to be modified.

mac_radio_defs.c
hal_board_cfg.h
mac_radio_def.h

closed 2020-08-18 by github-actions[bot]