Koenkk/Z-Stack-firmware #164: Details on how to set up a router

Gis70, 2020-04-28:
Hi,

First, thank a lot for this project.

I have some questions about the router.

I have a CC2531 flashed with the router firmware.

Can I plug this CC2531 in a standalone way ? (I would mean just to power it, no USB interface behind)

How to join the main network with this router ? need pairing with the coordinator ?

Maybe you can detail a bit more in the FAQ or write a chapter to set up a router.

Thansk you for these answer.

Garfonso, 2020-04-28:
You don't need to plug it into a working USB Port, just power is fine.

And yes, you need to pair it, allow devices to join and plugin power and it should join the network.

Gis70, 2020-04-28:
thanks, I was able to add the router by pairing it very close to the coordinator and i found a how to for router :)

closed 2020-04-28 by Gis70