Koenkk/Z-Stack-firmware #166: possible to disable router function on a bulb?

mgx0, 2020-05-11:
Hi, 

is there a possibility to disable routing function of a bulb? IKEA bulbs/outlets or Philips Hue (I have both). Problem is, that sometimes my bulbs are physically turned off and for a while I cannot communicate with devices that were connected through the bulb as a router

thanks

ptvoinfo, 2020-05-12:
@mgx0 It depends on the firmware. Usually, you cannot disable the router function.

mgx0, 2020-05-12:
Thanks a lot @ptvoinfo 

one more thing, if I may: imagine a bulb is connected and is acting as a router, there are devices using this router. Now I switch off the power to the bulb and the router disappears from the network. 

now the questions:
- Will the devices connect to another router automatically? or do I have to re-pair or force reconnect somehow?
- if automatically, how much time does it take?
- probably depends on a device, but do devices reconnect to another router/coordinator automatically when there's a better signal available? or are they stick to the device they were connected to, at any cost?

this will help me understand this amazing thing you people have created
thanks a lot

ptvoinfo, 2020-05-12:
1. Automatically.
2. Up to 24 hours (it depends on the device reporting interval).
3. Yes, they may reconnect to a router with a better signal.

mgx0, 2020-05-12:
thaaaaaaaaanks! good stuff!

closed 2020-05-12 by mgx0