Koenkk/Z-Stack-firmware #167: please explain the device limit

mgx0, 2020-05-11:
Hello,

sorry for asking stupid question, but I'm obviously not clever enough to understand.

If I flash regular (not a source routing one) coordinator firmware to CC2531 and router firmware on  two (2 devices) CC2530, how many devices can I connect?

will my devices communicate via connected routers or will the routers only consume a slot (therefore they are a waste dummy device)?

thanks a lot for explanation

mgx

ptvoinfo, 2020-05-12:
Hello,

Routers may help to increase the number of connected devices. But Zigbee devices search for a better path to a coordinator. The path is not predictable. All your devices may prefer a direct connection to the coordinator.

mgx0, 2020-05-12:
thanks @ptvoinfo 
I mean, I have a network of 25 devices at this moment.  I have been using default coordinator firmware on CC2531 and one router on CC2530. When looking onto network map, all devices were connected directly to Coordinator (including Router, but no devices were connected to Router). I have experienced some devices stopped communicating after some time (Xiaomi temperature sensors) and new devices could not be paired anymore (Ikea on/off switches). My question basically was, when using Default firmware, can I expect this behaviour (devices refuse to connect via router)? or is it just my devices have chosen not to connect to Router at all?

I have flashed Source Rooting firmware to Coordinator and now the map looks like it should look. 4 devices on Coordinator, the rest on the Router. So I'm considering adding another router to extend the range and to be on the safe side.

thanks a lot

mgx0, 2020-05-23:
just confirmed for myself that if I don't use the "source routing" firmware on main coordinator, devices are not using connected routers. As soon as I flashed "source routing" firmware, devices started to use routers

closed 2020-05-23 by mgx0