Koenkk/Z-Stack-firmware #168: Errors applying development Z-Stack_3.x.0 firmware.patch

jculbert, 2020-05-11:
There seems to be a problem with the development branch Z-Stack_3.x.0 firmware.patch for the znp.syscfg files. Attempting to apply the patch results in output the below. The content of the znp.syscfg files around line 50 is quite different from what is in firmware.patch. This is for the April 17 commit. The patch works for the April 13 commit.

error: patch failed: znp_CC1352P_2_LAUNCHXL_tirtos_ccs/znp.syscfg:52
error: znp_CC1352P_2_LAUNCHXL_tirtos_ccs/znp.syscfg: patch does not apply
error: patch failed: znp_CC2652RB_LAUNCHXL_tirtos_ccs/znp.syscfg:50
error: znp_CC2652RB_LAUNCHXL_tirtos_ccs/znp.syscfg: patch does not apply
error: patch failed: znp_CC26X2R1_LAUNCHXL_tirtos_ccs/znp.syscfg:49
error: znp_CC26X2R1_LAUNCHXL_tirtos_ccs/znp.syscfg: patch does not apply

Koenkk, 2020-05-12:
The update from 17 Apr assumes using Simplelink 4.10.00.78 instead of 3.40.00.02, make sure you are using the correct SDK.

jculbert, 2020-05-12:
Yes, I used Simplelink 4.10.00.78.

I investigated further and I see what's going on. Opening znp.syscfg (in the SysConfig editor) making some change, reverting the change, and then saving, completely changes the content (see below). It seems some kind of expansion is going on. If I do this for the three projects the patch file apply works fine. This wasn't needed for SDK 3.40.00.02.

Content of znp.syscfg at line 50 before saving:
var Power   = scripting.addModule("/ti/drivers/Power");
var RF      = scripting.addModule("/ti/drivers/RF");
var SHA2    = scripting.addModule("/ti/drivers/SHA2");
var TRNG    = scripting.addModule("/ti/drivers/TRNG");
var UART    = scripting.addModule("/ti/drivers/UART");
var zstack  = scripting.addModule("/ti/zstack/zstack");

Content at line 50 after saving:
CCFG.dioBootloaderBackdoor    = 13;
CCFG.levelBootloaderBackdoor  = "Active low";
CCFG.srcClkLF                 = "Derived from HF XOSC";
CCFG.ccfgTemplate.$name       = "ti_devices_CCFGTemplate0";

closed 2020-08-18 by github-actions[bot]