Koenkk/Z-Stack-firmware #169: Use CC1352P-2 like router

ozer59, 2020-05-14:
How to process to use CC1352P-2 like a router 

I have 2 CC1352P-2 1 for Coordinator i want use the second like routeur 
i try this firmware :https://github.com/Koenkk/zigbee2mqtt/issues/1429#issuecomment-581177622

 but i don't know how to process to pair it

Koenkk, 2020-05-14:
Currently there is no router firmware available. @ptvoinfo are you interested in building it?

ozer59, 2020-05-14:
yes a lot a need to extend my network because legrand router won't work with xiaomi devices sensor detector ...

ptvoinfo, 2020-05-15:
I don't have such a device and development tools for ARM. I cannot do it right now.

Koenkk, 2020-05-15:
@ptvoinfo would you be interested in building when we send you a CC2652R for free? (building tools are free btw)

ptvoinfo, 2020-05-15:
@Koenkk Sorry, it may take months.

gcormier, 2020-06-09:
Is the CC26x2R in the same situation? (No router support for unknown amount of time?)

asciidisco, 2020-06-25:
Is there any way to sponsor or support this?
I'm currently running a LAUNCHXL-CC26X2R1 coordinator & have 2 CC2531 running as routers & they don't really cut it. With 2 zig-a-zig-ah! devices sitting on my desk, I'm currently facing the decision to break my net up into 3 seperates ones (which I believe will come with a few other problems)...

So yeah, I'd be up for sponsoring this cause or support it as developer if there's something I can do.

gcormier, 2020-07-06:
I'd pitch in some money for router support for 1352P/CC2652R.

closed 2020-08-18 by github-actions[bot]

Pascal51882, 2020-10-05:
Is there still no firmware? For really big range networks it would be great :)