Koenkk/Z-Stack-firmware #172: What is the difference between source_routing and router firmware?

elbueno222, 2020-05-19:
I understand the difference between coordinator and router, but what is the difference between the source_routing firmware and router firmware?

- Source_routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing
- Router firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2531/bin

Aren´t they both router firmware for the same devices? If so why are they in different folders?

Koenkk, 2020-05-19:
The source routing firmware is a coordinator firmware, not a router firmware.

So comparing these makes no sense, instead you can compare the source routing vs the default firmware. 

The difference between them is explained here https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator#im-a-zigbee2mqtt-user-which-one-should-i-use and the overview above it.

elbueno222, 2020-05-19:
Ok, I have read those instructions like a hundred times and I just got it now, source routing is also a coordinator firmware but design for networks with more devices. It didn't make sense to me before because I was assuming the router firmware had the same limit of 5 direct children as the source routing but now I understand it doesn't make sense that it has this limit of 5 direct children.
Thank you very much, you can close this incident.

closed 2020-05-20 by Koenkk

pipiche38, 2022-05-13:
@Koenkk  not clear for me, but who is playing the role of collector ? Is that the coordinator as well ?

Koenkk, 2022-05-13:
@pipiche38 what do yo mean with "collector"?

pipiche38, 2022-05-13:
in the [reference](https://www.digi.com/resources/documentation/digidocs/90001537/references/r_large_zigbee_networks-source_routing.htm?TocPath=Working%20with%20Zigbee%7C_____14) 

> Source routing
> 
> In contrast to many-to-one routing that establishes routing paths from many devices to one data collector, source routing allows the collector to store and specify routes for many remotes. To use source routing on a network, many-to-one routes must first be established on the network from remote nodes to the central data collector.

Koenkk, 2022-05-13:
Yes it is the coordinator