Koenkk/Z-Stack-firmware #175: CC1352P2 after updating to the latest firmware network becomes unresponsive

cocooma, 2020-05-24:
Hi there,

After upgrading to zigbee2mqtt 1.13.1 and  the latest firmware I had repair all the devices. 
I'm just wondering am I missing some upgrade steps?

Thanks,
Oszkar

Koenkk, 2020-05-24:
Do you have a `coordinator_backup.json` in your zigbee2mqtt data directory? It should automatically restore that on the first start of a reflashed coordinator.

cocooma, 2020-05-24:
Yes I had it but the network was all over the place. Even after half and hour it was not sorting itself out. I have about 33 devices. Most of the are xiaomi light switches , door/motion sensors. With the most resent firmware I've only managed to pair 20 devices.

So I've ended up reverting it back to 20191106. 

Not sure what it the reason behind :/

Koenkk, 2020-05-25:
Do you have the debug logs when trying to control a plug after flashing the new firmware?

cocooma, 2020-05-26:
Yes I do:

In most cases it was complaining error is realted with no route to host:

 2020-05-24 11:05:40: Publish 'set' 'state' to 'SWS-Utility-Room' failed: 'Error: Command 0x001xxxxxxxxxxxxx/2 genOnOff.off({}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))'
debug 2020-05-24 11:05:40: Error: Command 0x001xxxxxxxxxxxxx/2 genOnOff.off({}, {"timeout":10000,"manufacturerCode":null,"disableDefaultResponse":false}) failed (Error: Data request failed with error: 'No network route' (205))
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:349:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)

Please let me know if you need anything else.

cocooma, 2020-05-26:
When I restored the previous firmware as well as the database, the state file and the coordinator_backup.json the network was still broken :/

I've ended up re-pairing everything

Koenkk, 2020-05-26:
`No network route' (205))` error after a reflash can most of the times be solved by power cycling the device.

iFloris, 2020-08-11:
Hello, I seem to have run into a very similar issue. I have been using this 1352p2 since January this year with great succes. 

However, during a three week absence, there was a power glitch and my home assistant (x64) machine rebooted, causing the adapter to become unavailable ([a known problem](https://github.com/Koenkk/zigbee2mqtt/issues/2162)). Reading about this problem to see if it can now be mitigated, I noticed there are both newer z-stack firmware as well as TI (XDS110) firmware. I updated both to the latest available versions last week.

Now, the network is at times slow to respond - a door sensor event causes a light to turn on after a very noticeable delay. Devices become unresponsive with time, with almost all my temperature sensors now offline. Restarting z2mqtt seems to help for a little while, but some devices don't come back. Repairing is also slow and sometimes does not show up in the log (with debug enabled) at all. 

In short, it quite feels like when I was using the CC2531 with 30 devices 😅

However, I think to have read that this device is among the most powerful and fast currently available, so I am hopeful that this is simply a matter of user error on my behalf or a mismatch between versions.

Do you have any suggestions in this matter @Koenkk ?

Koenkk, 2020-08-11:
@iFloris have you already tried with the latest firmware? When reflashing make sure  not to skip step 5 of https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html

iFloris, 2020-08-11:
Hello @Koenkk thanks for your reply, I really appreciate it. I see that you released new firmware just minutes ago (awesome!). I will give that a go right away and see how that works from tomorrow on.

iFloris, 2020-08-13:
Hello @Koenkk I have been testing the new firmware and things seem to be super awesome again. 🎉 
Thank you very much for the firmwares, the software and your unparalleled support.

Things respond as quickly as before and the network stabilised in about an hour time, which I think is very fast.
So the problem was likely with Coordinator 20200417 in conjunction with XDS110 3.0.0.12.

Current (awesomely) working version:

Version of Zigbee2Mqtt: 1.14.3 
Coordinator version: 20200805 
XDS110: 3.0.0.12

closed 2020-09-20 by github-actions[bot]