Koenkk/Z-Stack-firmware #176: Hosting Z-Stack 3.0 "end device" with router support firmware binaries?

Gamester17, 2020-05-25:
@Koenkk will you consider hosting Z-Stack 3.0 "end device" firmware with router support binaries?

* https://github.com/diyruz/diyruz_rt/releases
  * * https://github.com/diyruz/diyruz_rt

I think that many could be interested in third party Z-Stack 3.0 firmware for Sonoff BASICZBR3

* https://www.cnx-software.com/2020/05/25/zigbee-firmware-news-ti-z-stack-3-0-zigbee-for-cc2530-ptvo-zigbee-fw-configuration-tool/

@kirovilya Nice if Sonoff BASICZBR3 can be used as cheap Zigbee 3.0 sensor and router device?

Gamester17, 2020-05-25:
Please also see related requests https://github.com/Koenkk/Z-Stack-firmware/issues/153 and https://github.com/Koenkk/Z-Stack-firmware/issues/145

Koenkk, 2020-05-25:
I have nothing against this (if a proper manual is included how to flash the firmware/for which device).

ptvoinfo, 2020-05-26:
Hi @Gamester17,
I've tested a configurable firmware for Sonoff Basic ZBR3, but it is Zigbee 1.2

Gamester17, 2020-05-26:
@ptvoinfo Would you be willing to submit such a binary hex under a new "end_device" directory?

Gamester17, 2020-05-26:
@ptvoinfo Nice by the way! Did you happen to also test any external sensors with Sonoff BasicZBR3?

Would be great to know if it is compatible with any sensors that Itead sells for Sonoff TH10/TH16. Itead sells nicely packaged DS18B20, Si7021, and AM2301 sensors for its Sonoff TH10/TH16 series.

Would be awesome if those could be used with Sonoff BasicZBR3 with your third-party firmware.

Gamester17, 2020-05-26:
@kirovilya Are you willing to submit your Z-Stack 3.0 / Zigbee 3.0 binary for Sonoff BasicZBR3 as well?

ptvoinfo, 2020-05-26:
@Gamester17 My firmware requires several large companion binary files. I'm not sure that it is a good idea to upload all these files to the repository. But we can add a link to the corresponding info on my website.

closed 2020-08-18 by github-actions[bot]