Koenkk/Z-Stack-firmware #182: Best firmware for cc2538 +cc2592

SamuelCarson, 2020-06-16:
Many different firmware version are available.Very confusing.  Which is latest for MODKAM.RU V3 coordinator by Jager???

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:
cc2538 + cc2592

Koenkk, 2020-06-16:
@reverieline could you take a look?

reverieline, 2020-06-16:
> Many different firmware version are available.Very confusing. Which is latest for MODKAM.RU V3 coordinator by Jager???

The olny difference is connection interface.
MODKAMRU_V3_USB.hex for USB connected devices.
MODKAMRU_V3_UART-no-flow-control.hex for 4-wire UART connection (VCC,GND,TX,RX).
MODKAMRU_V3_UART-flow-control.hex for 6-wire UART connection (VCC,GND,TX,RX,RTS,CTS).

SamuelCarson, 2020-06-16:
I had found these and they seemed to be to latest versions. BUT there a many others referred to in the CC2538 discussions. I just wanted to be sure these were the latest available.  Thanks

From: reverieline 
Sent: Tuesday, June 16, 2020 1:31 PM
To: Koenkk/Z-Stack-firmware 
Cc: SamuelCarson ; Author 
Subject: Re: [Koenkk/Z-Stack-firmware] Best firmware for cc2538 +cc2592 (#182)

  Many different firmware version are available.Very confusing. Which is latest for MODKAM.RU V3 coordinator by Jager???

The olny difference is connection interface.
MODKAMRU_V3_USB.hex for USB connected devices.
MODKAMRU_V3_UART-no-flow-control.hex for 4-wire UART connection (VCC,GND,TX,RX).
MODKAMRU_V3_UART-flow-control.hex for 6-wire UART connection (VCC,GND,TX,RX,RTS,CTS).

closed 2020-08-18 by github-actions[bot]