Koenkk/Z-Stack-firmware #184: Enabling multiprotocol (with DMM: BLE+Zigbee) on CC26x2

danergo, 2020-06-21:
Hi,

I think it would be great to collect here all our common efforts to activate the DMM feature on the CC2652. As there is also a convenient and not-expensive Launchpad available, putting together the things shall not be super hard.

I've just registered for TI's resources (already have the Launchpad), but if someone already has done some progress, please share it, so we could continue together.

MPM1107, 2020-06-22:
Just my two cents with the little knowledge I have but I feel like using DMM on a zigbee coordinator seems like a terrible idea. The whole zigbee network expects the coordinator to be available 100% of the time, so if the 2652 is talking over BLE right as a zigbee message is sent to it that would probably result in the end device that sent the message dropping out of the network, maybe even the whole network collapsing.

danergo, 2020-06-22:
I have the same fear. DMM seems to be suitable for being end device and BLE peripherial.

But what is the right methodology then to be able to use both ZB Coordinator and BT central on the same board? Two separate chips with two separate antennas placed well above the λ/4 distance from each other? Even we do that, results would terrible as those two will fight for the channel, no?

Garfonso, 2020-06-22:
Bluetooth does not fight about channels, but hops frequencies, i.e. will disturb all "fixed frequency" networks (like zigbee &  wifi) only for a short time and will be disturbed by them only for a short time interval.

Not sure if the fear of blocking zigbee messaging with ble functionality is rational or not. Maybe there is some solution for that?

My main question would be what advantage you hope for with adding BLE to the same board? From what I see bluetooth receiver capable of BLE are very common & cheap (and even included in Raspberry PI for example). So I do not really see the need here.

danergo, 2020-06-22:
I have some CC2652 + CC2591 frontend, which can be a very decent and stable ZBC, and I also have a custom board without BT. 

My intention is to integrate both BT and ZB, and I thought with this single SoC I can do it (even TI confirmed it long time ago, but it was not a precise question from my end: DMM is only for BLE peripherial + ZB*).

You are right though, BLE receivers are common, but then I would need two separate modules (or at least chips), and two separate antennas. I think two separate modules can work on the same antenna, but you have to be an antenna-and-protocol expert to do that.

Now, as my CC2652 module is equipped with CC2591, it increases the efficiency a bit, it also has external antenna connector => it wouldn't be too hard to put my board into a case.
With separate BT module things getting (over)complicated.

danergo, 2020-06-22:
By the way, here are TI's resources for the DMM: https://software-dl.ti.com/simplelink/esd/simplelink_cc13x2_26x2_sdk/3.40.00.02/exports/docs/dmm/dmm_user_guide/html/dmm/performance.html

They state here under limitations: "The DMM has only been tested/characterized with the stack and role combinations covered by the examples provided in the SimpleLink CC13x2 / 26x2 SDK."

And none of their example mentions BLE Central functionality => I assume they have not "tested/characterized" their solution for this.

closed 2020-08-18 by github-actions[bot]