Koenkk/Z-Stack-firmware #185: [Z-Stack 3.0.2] Compile problem for cc2530_CC2592

Boia11, 2020-06-25:
Hi,
I try to compile the Z-Stack firmware v3.0.2 coordinator for a cc2530_CC2592. I use latest IAR v10.30.1.
With Stack Size/XDATA: 0x400 return error....I try 0x390 compile hex file but after flash the chip not responding (with your [hex](coordinator/Z-Stack_3.0.x/bin/CC2530_CC2592_20190523.zip) from repository work fine).
This is the error:
```
Linking 
Error[e16]: Segment XDATA_I (size: 0xc1 align: 0) is too long for segment definition. At least 0x57 more bytes needed. The problem occurred while processing the segment placement command  
"-Z(XDATA)XDATA_Z,XDATA_I=_XDATA_START-_XDATA_END", where at the moment of placement the available memory ranges were "XDATA:1e96-1eff" 
   Reserved ranges relevant to this placement: 
   XDATA:1-400          XSTACK 
   XDATA:401-1eff       XDATA_Z 
   BIT:0-7              BREG 
   BIT:80-87            SFR_AN 
   BIT:90-97            SFR_AN 
   BIT:a0-af            SFR_AN 
   BIT:b8-c7            SFR_AN 
   BIT:f8-ff            SFR_AN 
Error while running Linker 
 
Total number of errors: 1 
Total number of warnings: 486 
```
This other recorsive warning messages is normal?
```
Warning[Pe047]: incompatible redefinition of macro "FAKE_CRC_SHDW" C:\Texas Instruments\Z-Stack 3.0.2\Projects\zstack\ZNP\Source\preinclude.h 4 
Warning[Pe047]: incompatible redefinition of macro "TC_LINKKEY_JOIN" C:\Texas Instruments\Z-Stack 3.0.2\Projects\zstack\ZNP\Source\preinclude.h 8 
Warning[Pe047]: incompatible redefinition of macro "ASSERT_RESET" C:\Texas Instruments\Z-Stack 3.0.2\Projects\zstack\ZNP\Source\preinclude.h 9 
Warning[Pe047]: incompatible redefinition of macro "ZDSECMGR_TC_DEVICE_MAX" C:\Texas Instruments\Z-Stack 3.0.2\Projects\zstack\ZNP\Source\preinclude.h 13 
Warning[Pe047]: incompatible redefinition of macro "DISABLE_GREENPOWER_BASIC_PROXY" C:\Texas Instruments\Z-Stack 3.0.2\Projects\zstack\ZNP\Source\preinclude.h 14 
Warning[Pe047]: incompatible redefinition of macro "MAXMEMHEAP" C:\Texas Instruments\Z-Stack 3.0.2\Projects\zstack\ZNP\Source\preinclude.h 93 
```

Koenkk, 2020-06-29:
I have compiled this firmware with:
- XData:
[image]
- Config: CC2530 - ZNP-Without-SBL
- Defined symbols only: `FIRMWARE_CC2530_CC2592` (make sure to clear others!! Otherwise you will get the warnings shown in your post)
- Iar workbench 8051 10.20.1
- If compilation fails due to not enough space, reduce the `MAXMEMHEAP` in `preinclude.h` a or `XDATA` a bit.

Boia11, 2020-06-30:
>     * Defined symbols only: `FIRMWARE_CC2530_CC2592` (make sure to clear others!! Otherwise you will get the warnings shown in your post)

Thanks @Koenkk , but I think is better explicit that in the compile.md file :sweat_smile:

closed 2020-06-30 by Boia11

Koenkk, 2020-07-01:
Updated the instructions, thanks