Koenkk/Z-Stack-firmware #186: E18-MS1PA1-PCB Does not work after flashing (CC2530+CC2592)

dstasiuk, 2020-06-29:
Hi guys.
This EBYTE E18-MS1PA1-PCB module does not work after flashing. 
I try 2 different modules and the result is the same - after reflashing, it does not start.
Connected GND, 3.3V, P2.1, P2.2, RST, successfully flash, and pass the verify.
[image]

dstasiuk, 2020-06-29:
I need connect P2.0 to GROUD?

codmpm, 2020-07-07:
Hey @dstasiuk,

here's the wiring of my [E18-MS1PA1-IPX Raspberry Pi UART Adapter](https://shop.codm.de/automation/zigbee/15/zigbee-long-range-raspberry-pi-funkmodul-v0.3).

[image]

P0.2 is RDX, thus TXD on the Pi
P0.3 is TXD, thus RXD on the Pi

Only VCC and GND are needed for the function. 
For programming you DC/DD, which are P2.2/P2.1, as well as RST, GND and VCC.

I flashed CC2530_CC2592_20190523.zip from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin for Z-Stack 3.0.x or CC2530_CC2592_DEFAULT_20190608.zip from https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default as a coordinator.

Cheers,
Patrik

closed 2020-07-13 by dstasiuk