Koenkk/Z-Stack-firmware #188: AC Powered CC2530  + CC2591 router

lealoureiro, 2020-07-07:
hi Guys,

Just built my how AC Powered CC2530  + CC2591 following the tutorial below:

https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html

But when is very quiet i can hear a very light buzz coming from the box. Is this normal?

To be honest I'm not very confident connecting the HLK-PM03 to main power socket? is that safe? what can go wrong like catch fire or short circuit?

I've seen some guys building a breakout board around it with fuses and so, would this not be a more reliable/safe solution? 

Thanks in advance!

Leandro

Koenkk, 2020-07-07:
I'm definitely no electrician, but I consider the HLK-PM03 to be safe, already running my CC2530 router 1.5 years with this.

lealoureiro, 2020-07-08:
okay, good to know because in fact the reliability so far is much better

closed 2020-08-18 by github-actions[bot]