Koenkk/Z-Stack-firmware #192: CC2538 + CC2592 Issues

michalszymura, 2020-07-15:
So i just flashed CC2538 + CC2592 with CC2538_CC2592_MODKAMRU_V3.
I'm on dev branch of zigbee2mqtt
I discovered that after the device looses power it's unable to reconnect to any of my bulbs. Weirdly hue remote works fine.
I have to repair all the bulbs. Not sure what's the problem.

michalszymura, 2020-07-16:
I can confirm same behaviour after removing pan id and extended pan id, removing all databases and re-flashing the stick. Not sure what else i could try?

michalszymura, 2020-07-16:
It somehow started working properly when i tried starting container without stick plugged in and then with stick plugged in it reconnected to everything.

kirovilya, 2020-07-16:
@michalszymura try previous version firmware (20200211) https://github.com/reverieline/CC2538-CC2592-ZNP/tree/967b3b0ce3302a2dda1ec6f4cd181c0f086dda01

michalszymura, 2020-07-16:
The older firmware works fine.

closed 2020-08-11 by michalszymura