Koenkk/Z-Stack-firmware #193: Z-Stack_Home_1.2 source_routing for CC2538 + CC2592

falmanna, 2020-07-25:
Hi,

I am wondering if the provided `CC2530_CC2592_SOURCE_ROUTING_20190619.zip` will work with CC2538 + CC2592.
The default `1.2.2a` firmware provided by TI does work on all cc253x chipsets, so theoretically this should wok unless there are some memory specific configs. 

If won't work, do you plan on releasing a `1.2 source routing` firmware for the CC2538 + CC2592?
Thanks.

Koenkk, 2020-07-25:
No this wont work as its a completely different chip. Note that the existing 3.0 firmware is already a source routing one.

falmanna, 2020-07-25:
Thanks @Koenkk 
Actually it is from the same family of chips and the default firmware is identical for all cc253x chips. But since it is a modified version I guess it might have some hardware-specific configs,  which will break the compatibility.
I've seen the 3.0 one, but I was looking for 1.2 version.

closed 2020-07-25 by falmanna