Koenkk/Z-Stack-firmware #195: Adapter fails almost every day.

likellindil, 2020-08-09:
Host: NUC i3 running HA with Z2M/Z2MA and Node-red add-ons via VMware.
Coordinator: CC26X2R1_20200805
Routers: CC2530_CC2591_router_2019_02 x 2
Devices: All Xiaomi and Aqara devices, including covers, wall switches, power switches, motion sensors, door & window sensors, and water leak sensors. Around 100 devices in one Zigbee network.

It works well the first several hours after being configured.
However, some of those devices become unresponsive randomly after that.
Almost every day, when I get up in the morning, the network is dead and all devices are unresponsive.

This is what I do to fix this temporarily. 
1. Unplug the coordinator from NUC.
2. Re-plug it
3. Restart the Z2M add-on in HA
Most of the devices will back alive one by one in about half an hour.
4. For the rest devices, I need to permit join in Z2MA, and then re-pair them by pressing and holding the pairing keys. 

I have also tried those solutions mentioned on the Z2M website, forum, and issues on Github, none of them works.
A. I do press the reset key next to the USB port after plugging the coordinator to NUC, it doesn't change anything.
B. I erased the coordinator board and flash it with the latest master and develop firmware, none of them works, the develop version is more unstable.
C. I have removed those IKEA devices from the network ( they used to be part of it ), with only Xiaomi and Aqara devices left there, no progress.
D. I tried to remove those routers from the network and re-add them in, it doesn't help.

Here is the log after the coordinator is dead:
2020-08-09T13:54:30: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-08-09T13:54:30: PM2 log: App [npm:0] starting in -fork mode-
2020-08-09T13:54:30: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.2 start /zigbee2mqtt-1.14.2
> node index.js
zigbee2mqtt:info  2020-08-09 13:54:30: Logging to console and directory: '/share/zigbee2mqtt/log/2020-08-09.13-54-30' filename: log.txt
zigbee2mqtt:info  2020-08-09 13:54:31: Starting zigbee2mqtt version 1.14.2 (commit #unknown)
zigbee2mqtt:info  2020-08-09 13:54:31: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-08-09 13:54:50: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-08-09 13:54:50: Failed to start zigbee
zigbee2mqtt:error 2020-08-09 13:54:50: Exiting...
zigbee2mqtt:error 2020-08-09 13:54:50: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.14.2/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:87:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.2/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno
 1
npm ERR! zigbee2mqtt@1.14.2 start: `node index.js`
npm ERR!
 Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.2 start script.
npm
 ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-08-09T05_54_50_708Z-debug.log

Koenkk, 2020-08-09:
Do you have a log at the moment it crashes?

likellindil, 2020-08-10:
I just checked the log folder, it looks like this:
info  2020-08-09 17:41:05: Logging to console and directory: '/share/zigbee2mqtt/log/2020-08-09.17-41-05' filename: log.txt
info  2020-08-09 17:41:06: Starting zigbee2mqtt version 1.14.2 (commit #unknown)
info  2020-08-09 17:41:06: Starting zigbee-herdsman...
error 2020-08-09 17:41:25: Error while starting zigbee-herdsman

I have set the log level to warning and above, I will upload a copy next time it crashes.

likellindil, 2020-08-13:
It fails again last night.
However, there are 10 folders in the log folder, all of them only have the same record like I posted above.
Is there any way to remove the quantity limit of the log records?

Koenkk, 2020-08-13:
Changing in the systemctl config (https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#5-optional-running-as-a-daemon-with-systemctl) `Restart=always` to `Restart=no` should prevent the restarts (and you will only end up with 1 log dir)

likellindil, 2020-08-14:
> Changing in the systemctl config (https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#5-optional-running-as-a-daemon-with-systemctl) `Restart=always` to `Restart=no` should prevent the restarts (and you will only end up with 1 log dir)

Thanks.

It fails again today and I got these:
1. Before the adapter was down, all devices were unresponsive. The log shows something like this:
`info  2020-08-14 05:36:00: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b001caa6ab0","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack3x0","dateCode":"20200805","lastSeen":1597354560090},{"ieeeAddr":"0x00158d00047b9965","type":"EndDevice","networkAddress":61171,"model":"RTCGQ11LM","vendor":"Xiaomi","description":"Aqara human body movement and illuminance sensor","friendly_name":"MS_DIN_R","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_motion.aq2","hardwareVersion":1,"softwareBuildID":"3000-0001","dateCode":"20170627","lastSeen":1597346207897},{"ieeeAddr":"0x00124b001f320e8e","type":"Router","networkAddress":9873,"model":"CC2530.ROUTER","vendor":"Custom devices (DiY)","description":"[CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/)","friendly_name":"RT_DR","manufacturerID":4447,"manufacturerName":"LUMI","powerSource":"Mains (single phase)","modelID":"lumi.router","hardwareVersion":1,"dateCode":"2018-09-26","lastSeen":1597347234491},{"ieeeAddr":"0x00158d000363e359","type":"EndDevice","networkAddress":2564,"model":"QBKG04LM","vendor":"Xiaomi","description":"Aqara single key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter","friendly_name":"LS_GR","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"DC Source","modelID":"lumi.ctrl_neutral1","hardwareVersion":38,"dateCode":"04-25-2018","lastSeen":1597347260047},{"ieeeAddr":"0x00158d00044c3b2e","type":"EndDevice","networkAddress":22696,"model":"WXKG12LM","vendor":"Xiaomi","description":"Aqara wireless switch (with gyroscope)","friendly_name":"WBTN_GR","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_switch.aq3","hardwareVersion":1,"softwareBuildID":"3000-0001","dateCode":"20180629","lastSeen":1597345329976},{"ieeeAddr":"0x00158d0003f3a1a8","type":"EndDevice","networkAddress":65354,"model":"RTCGQ11LM","vendor":"Xiaomi","description":"Aqara human body movement and illuminance sensor","friendly_name":"MS_GR","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_motion.aq2","hardwareVersion":1,"softwareBuildID":"3000-0001","dateCode":"20170627","lastSeen":1597342493521},{"ieeeAddr":"0x00158d0003fb5c34","type":"Router","networkAddress":33196,"model":"ZNCLDJ11LM","vendor":"Xiaomi","description":"Aqara curtain motor","friendly_name":"CT_GR","manufacturerID":4447,"manufacturerName":"LUMI","powerSource":"Mains (single phase)","modelID":"lumi.curtain","hardwareVersion":17,"dateCode":"08-13-2018","lastSeen":1597347015832},{"ieeeAddr":"0x00158d0003599d0f","type":"EndDevice","networkAddress":1723,"model":"QBKG04LM","vendor":"Xiaomi","description":"Aqara single key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter","friendly_name":"LS_ER","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"DC Source","modelID":"lumi.ctrl_neutral1","hardwareVersion":38,"dateCode":"04-25-2018","lastSeen":1597347248620},`
It looks like the coordinator was trying to get all the devices back but none of them responsed.

2. After a period of time, the adapter itself was dead:
`error 2020-08-14 09:01:49: Adapter disconnected, stopping`

3. I noticed that the board was very hot when I tried to unplug and re-plug it. 

4. After re-plug, it gradually got all those devices back.

Koenkk, 2020-08-14:
Can you provide the complete debug logging of this?

> It looks like the coordinator was trying to get all the devices back but none of them responsed.

Based on what do you conclude this?

likellindil, 2020-08-14:
[log.txt](https://github.com/Koenkk/Z-Stack-firmware/files/5074619/log.txt)
The attachment is one of the log file

likellindil, 2020-08-14:
> Can you provide the complete debug logging of this?
> 
> > It looks like the coordinator was trying to get all the devices back but none of them responsed.
> 
> Based on what do you conclude this?

The attachment is one of the log files

[log.txt](https://github.com/Koenkk/Z-Stack-firmware/files/5074626/log.txt)

dzungpv, 2020-08-16:
This is your problem:
```
Routers: CC2530_CC2591_router_2019_02 x 2
```
I have bad experience with all other router except cc2652/1352 with latest sdk(3.4 or later).

My network:
1, Xiaomi sensors, only end device, not router like curtain.
2, Router: CC2652
3, ZC: cc1352p

likellindil, 2020-08-16:
> cc2652/1352

Thanks.
It will be appreciated if you could help me with another two questions:
1. Is it okay to use CC26X2R1 as the coordinator and use CC2652/1352 as routers?
2. Where could I find the router firmware for  CC2652/1352?

dzungpv, 2020-08-17:
> 
> 
> > cc2652/1352
> 
> Thanks.
> It will be appreciated if you could help me with another two questions:
> 
>     1. Is it okay to use CC26X2R1 as the coordinator and use CC2652/1352 as routers?
> 
>     2. Where could I find the router firmware for  CC2652/1352?

What channel are you running?

dzungpv, 2020-08-17:
I just update router firmware for CC2652R1 dev board, you can download it here: https://github.com/dzungpv/Z-Stack-firmware/tree/master/router/CC2652R1

likellindil, 2020-08-19:
> I just update router firmware for CC2652R1 dev board, you can download it here: https://github.com/dzungpv/Z-Stack-firmware/tree/master/router/CC2652R1

Thanks!
Just got time to work on it. 
The channel is 11.

So I just need to purchase another LAUNCHXL-CC26X2R1 board, flash the router firmware and use it to replace those 2 CC2530 routers, right?

Any other recommendations? The LAUNCHXL-CC26X2R1 board is too big to be used as a router anyway...

I also talked with Sascha, the creator of the slaesh’s CC2652RB stick, it is much more compact. 
However, he doesn't have a firm schedule for a router firmware. 

Thanks again for your help.

Best regards,
Mandarin Duck

dzungpv, 2020-08-25:
> Thanks!
> Just got time to work on it.
> The channel is 11.
>
> So I just need to purchase another LAUNCHXL-CC26X2R1 board, flash the router firmware and use it to replace those 2 CC2530 routers, right? [...]

Use channel 26 and plan your wifi channel with lowest interference. For the CC2652RB, i can build it in free time, but in order to flash it, you must have a programmer, TI kit include programmer so it not the problem.

zen2, 2020-09-01:
If you need a cheap router, I'm really happy with this kind of devices:
https://fr.aliexpress.com/item/32966584298.html

I got actually 30 end devices connected with it and I found it even more reliable than HUE bulbs.

dgomes, 2020-09-01:
> If you need a cheap router, I'm really happy with this kind of devices:
> https://fr.aliexpress.com/item/32966584298.html
> 
> I got actually 30 end devices connected with it and I found it even more reliable than HUE bulbs.

Any of those devices are xiaomi/aqara devices ? They seem to be picky about connecting to routers...

zen2, 2020-09-01:
> Any of those devices are xiaomi/aqara devices ? They seem to be picky about connecting to routers...
It's mainly xiaomi/aqara devices ! That's why I advice it. The only thing that I don't know if it's a router that support zigbee 3.x.

likellindil, 2020-09-06:
> Use channel 26 and plan your wifi channel with lowest interference. For the CC2652RB, i can build it in free time, but in order to flash it, you must have a programmer, TI kit include programmer so [...]

Thanks! Just got time to make progress on my HA setup.
I have ordered several slaesh’s CC2652RB sticks, it will spend another 20 days or even more on the way so I guess I should just wait.
I will also try the cheap router recommended by zen2 as well.

likellindil, 2020-09-06:
> > Any of those devices are xiaomi/aqara devices ? They seem to be picky about connecting to routers...
> 
> It's mainly xiaomi/aqara devices ! That's why I advice it. The only thing that I don't know if it's a router that support zigbee 3.x.

Thanks!
I will get some from our local market.
My coordinator is working under Zigbee 3.X.
I will post here if the router works well with it.

closed 2020-10-14 by github-actions[bot]