Koenkk/Z-Stack-firmware #196: Is it possible to change the pan-id of cc2531 usb-dongle

Bhavana-S, 2020-08-11:
Hi,
  
First of all i am beginner to the cc2531 module.
  
As specified in the link https://www.zigbee2mqtt.io/getting_started/flashing_the_cc2531.html i have flashed cc2531 usb-dongle with Z-Stack_Home_1.2 coordinator firmware using https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default using Flash Programmer tool of TI.

Also i have flashed a router from https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2531/bin to another cc2531 usb-dongle.

I have checked the interaction of data from co-ordinator to router(end_device). 

My question is if i have to make co-ordinator and end-device to work the same in another zigbee network i have to change the pan-id. Is there a way to change the pan-id of the coordinator and router(end-device). 
According to my requirement i have to change the pan-id, is it possible to change the pan-id if cc2531 usb-dongle is flashed with above firmware?
If so what is that? 
If i can which tool has to be used to change the pan-id?
If not why is it not possible to change the pan-id?

Thanks in advance

Koenkk, 2020-08-11:
Then panID can be written on the first start of the stick or after a reset. Example in zigbee-herdsman: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/z-stack/adapter/startZnp.ts#L166

closed 2020-08-11 by Koenkk

Bhavana-S, 2020-08-17:
Thank you @Koenkk  for the reply. Now i am able to change the pan-id of the coordinator.

i have flashed end-device firmware https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2531/bin with this. 

Then how can i change the pan-id of cc2531 end-device/router.