Koenkk/Z-Stack-firmware #199: Coordinator network commissioning and some other stuff

aromanro, 2020-08-19:
Hi,

I already mentioned the issue here: https://github.com/Koenkk/zigbee2mqtt/issues/1445 but it turns out that it's more than that. I actually have two devices now with a 3.0.x firmware from here that exhibit the same thing.

One is an usb stick with cc2531, one is a serial device for raspberry with CC2530+CC2592.
They seem to work all right so far (maybe with some issues I'm still investigating), but there is one thing that does not work in both cases: Trying to commission a new network. I'm using my own code to communicate with the coordinator devices.

The app cnf bdb start commissioning request (mode: 0x4 - BDB_COMMISSIONING_MODE_NWK_FORMATION) failes with an app cnf bdb commission notification with status 0x8 (BDB_COMMISSIONING_FORMATION_FAILURE).

I tried various settings for the channels mask, setting a particular pan id or letting it unset (that is, to be randomly generated), with the same result.

Is there something I'm missing when doing network commissioning?
 
The usb stick also has a led which lights up when the stick is inserted and it stays that way for a while. I don't know what it's doing during that time, but while it's lit, communication with the stick is limited, it doesn't reply to reset commands or to 'sys ping'.

Koenkk, 2020-08-19:
The zigbee-herdsman commissioning code can be found here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/z-stack/adapter/startZnp.ts#L157

Note that this is not the correct place for z-stack firmware support, I recommend using: https://e2e.ti.com/

closed 2020-08-19 by Koenkk

aromanro, 2020-08-19:
Thank you!

The generic z-stack support is indeed there, but I was addressing the specific firmwares found here, as there are a lot of options for compiling which can affect a lot the behavior.

I also have firmwares compiled by myself, including for serial cc2530 for raspberry (also for cc2538) which do not exhibit this behavior.