Koenkk/Z-Stack-firmware #200: [Feature request] E18-MS1PA2-IPX / E18-MS1PA2-PCB support

papperone, 2020-08-19:
Just come across these new modules from eByte which don't use a standard CC2592 amplifier but a new one which is built-in (accordin to the datasheet) and wonders if someone knows if they can be flashed and which firmware works in order to be able to use the PA correctly
links:
http://www.ebyte.com/en/product-view-news.aspx?id=919
http://www.ebyte.com/en/product-view-news.aspx?id=988

papperone, 2020-09-19:
nobody is able to help here?
If I will purchase some of these for test how I can make them work with ZStack ?

Boia11, 2020-09-30:
Look #183

closed 2020-11-07 by github-actions[bot]

lexxnsk, 2021-01-11:
Also interested in solution for pa2 module

papperone, 2021-02-01:
why this request is completely unattended?
it's proved that this module requires diffrerent pin to enable the PA compared with CC2591, so a new compile is required.
here you can find a router version which is working perfectly reaching same performance of CC2530+CC2592
https://ptvo.info/zigbee-configurable-firmware-features/e18-ms1pa2-ipx/

can someone create a proper coordinator firmware for this module??

thanks!

codmpm, 2021-02-11:
As a customer still needs Z-Stack 1.2 support for his special application and ebyte will likely discontinue the PA1, I'm also interessted in a coordinator firmware for the PA2.
My ebyte sales representative told me that the PA1-PCB is no longer on sale. Luckily I still could get some more PA1-IPX, though.

Koenkk, 2021-02-11:
I don't plan to do any further development for the CC2530/CC2531 (expect small bug fixes) since my focus is on the CC2652/CC1352 now.

papperone, 2021-02-11:
> 
> 
> I don't plan to do any further development for the CC2530/CC2531 (expect small bug fixes) since my focus is on the CC2652/CC1352 now.

hi Koen, may I ask you if you can make a quick and dirty firmware for this module? as far as I understood it's only a matter of using different GPIO for the PA as per the link I've sent you before (cc2591 version does work but does not acitvate the PA, so the range is quite poor)... I won't call it "development" but rather then adapting to a hardware change as the old module can't be procured anymore!

lexxnsk, 2021-02-12:
> hi Koen, may I ask you if you can make a quick and dirty firmware for this module? as far as I understood it's only a matter of using different GPIO for the PA as per the link I've sent you before [...]

Agree. It would be great to adapt it

Koenkk, 2021-02-12:
@papperone since you have this module I suggest that you try to make the required changes first. Instructions on how to compile the firmware can be found here: https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/COMPILE.md . If you got it working please share what had to be changed then I can look to incorporate them.

papperone, 2021-02-13:
> @papperone since you have this module I suggest that you try to make the required changes first. Instructions on how to compile the firmware can be found here: [...]

I've never done this... I can give it a try and let you know... I thougth as you have all setup that was a matter of 10mins to recompile with new GPIO settings; if you can do it and send it to me only the firmware I can test it as I do have those modules in-house, if not possible thanks anyway, will see what can I do by myself...

codmpm, 2021-02-13:
@papperone Which GPIO's have to be changed? Maybe you can share the info or point me where I can find it.

papperone, 2021-02-13:
@codmpm I posted the lin before --> https://ptvo.info/zigbee-configurable-firmware-features/e18-ms1pa2-ipx/
here they have only the router firmare for the PA2 modules, it is reported that the correct PINs are TXEN=P11 and RXEN=P10
if you have the skills to recompile the coordinator firmware I'll be interesting to have it to test on the PA2 modules (if you agree of course), thanks!

codmpm, 2021-02-13:
We are currently working out some problems with the CC2538 module and learn to compile the coordinator firmware by ourselfs. Hopefully the PA2 firmware is another outcome of this...
But neverthelss, would be great if more can jump in.

papperone, 2021-02-20:
> @papperone since you have this module I suggest that you try to make the required changes first. Instructions on how to compile the firmware can be found here: [...]

Hi Koen, the issue is not how to recompile the firmware (this I thin I can manage as the procedure is quite clearly explained) but where to change the IOs :( I've read carefully the patch code and can't find any reference to  RF RX/TX IOs to update :( SIGH!

papperone, 2021-02-21:
> @papperone since you have this module I suggest that you try to make the required changes first. Instructions on how to compile the firmware can be found here: [...]

hi Koen, the issue it not recompiling the firmware, this looks quite a striaght process, but despite I read your patches multiple times I can't find any reference to the IOs used to activate the PA, I would apreciate any hints here due to your high experience on this firmware... :)

emc02, 2021-04-27:
@Koenkk I would also be very interested to make these modules work. This page (https://ptvo.info/zigbee-configurable-firmware-features/e18-ms1pa2-ipx/) says, that P11 is TXEN and P10 is RXEN, but I have not the experience to change that in any way on your files. But I have such modules (E18-MS1PA2-PCB) at home and would test the changes if you can do a quick and dirty version of it.

Kugeleis, 2021-06-19:
Is there progress on this issue?

papperone, 2021-06-20:
it seems nobody is interested in making this module works; pity as the only ones have been phased out and only the new ones are available on the market; on top there's a new version with even a better PA (27db) but I guess it will be the same story :(

DangLhb, 2025-01-15:
> it seems nobody is interested in making this module works; pity as the only ones have been phased out and only the new ones are available on the market; on top there's a new version with even a better PA (27db) but I guess it will be the same story :(

Hello. Is your module working yet? I bought it and don't know how to use it!!!