Koenkk/Z-Stack-firmware #201: cc2530+2591 zip file not valid?

sander816, 2020-08-26:
i am trying to download https://github.com/Koenkk/Z-Stack-firmware/blob/master/router/CC2530_CC2591/bin/CC2530_CC2591_router_2019_02.zip

but i cannot unpack it:
pi@rpi3:~/flash_cc2531$ unzip CC2530_CC2591_router_2019_02.zip
Archive:  CC2530_CC2591_router_2019_02.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of CC2530_CC2591_router_2019_02.zip or
        CC2530_CC2591_router_2019_02.zip.zip, and cannot find CC2530_CC2591_router_2019_02.zip.ZIP, period.

are there othert resources for router firmware?

Koenkk, 2020-08-26:
Works fine for me, what's the size of the downlaoded zip?

sander816, 2020-08-27:
Sorry, my fault...

closed 2020-08-27 by Koenkk