Koenkk/Z-Stack-firmware #203: No LED feature on zzh

elfhack, 2020-08-31:
# Bug Report
## What happened
LED is not on on zzh, even when disable_led: false

## What did you expect to happen
LED would be on

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.14.3
Adapter hardware: CC26X2R1
Adapter firmware version: CC26X2R1_20200805

omerk, 2020-09-15:
Just for reference, all Electrolama boards (current: zzh, future: zzh-p and zoe2) have an LED on `DIO_7` (active high).

omerk, 2020-10-16:
@Koenkk Looking at the TI ZNP docs, perhaps this is what can be used for implementing this?

[image]

Have not yet looked into what changes in the firmware this will need but at least there is provision for GPIO control in ZNP commands.

Koenkk, 2020-10-16:
@omerk I prefer to keep it under the led control command, and modify the implementation of the led control command in the firmware. Zigbee-herdsman shouldn't be bothered with such details.

But in order to do this:
- LED for supported devices should always be under `DIO_7`, OR:
- We should be able to detect a ZZH from the firmware

Sloth-on-meth, 2020-11-17:
Still relevant!

Sloth-on-meth, 2020-12-18:
Still relevant

Sloth-on-meth, 2021-01-18:
Still relevant

njordan77, 2021-02-02:
Also relevant for myself. would also like to see the LED as the status.

omerk, 2021-03-05:
This is very much a cosmetic improvement but commenting to keep this issue open.

Looks like we managed to agree on `DIO_7` being the standard LED on all the current boards, going with the hardware details [here](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/README.md).

Koenkk, 2021-03-18:
Thanks to @egony I've got the leds working.

Next thing is to decide on what to do with them. My current idea is to turn them on when permit joining is enabled (unless `disable_led: false` is set in the `configuration.yaml`.

egony, 2021-03-19:
In my firmware LED blinking when join enabled and turns on when network started. Another LED flashed when packet recieved.
See readme here:
https://github.com/egony/cc2652p_E72-2G4M20S1E/tree/master/firmware/coordinator

Gio-dot, 2021-03-19:
> In my firmware LED blinking when join enabled and turns on when network started. Another LED flashed when packet recieved.
> See readme here:
> https://github.com/egony/cc2652p_E72-2G4M20S1E/tree/master/firmware/coordinator

Tried yesterday on my adapter: it is a very useful use of the leds. In this way leds gives very useful info.

Gio-dot, 2021-03-19:
For  all interested in leds use: i think best use of 2 leds would be:
led1= working or pairing
led2= packet received

Assuming this is the right led use: it would be possible to add a simple led configuration in z2m experimental settings?
Example:
Running/pairing led DIO: DIO_8
Packet received led DIO: DIO_7

What do you think about?

omerk, 2021-03-19:
As a first step, we can assume that `DIO_7` is present on most boards therefore it could be used as the one LED to be used as the "join enabled" indicator. Perhaps the second LED and configurable-IO can be added later on but I'm a bit worried that it just might be unnecessary complexity that can confuse people. If there is indeed interest in extra LEDs etc, my vote would be to have them turned off with explicit configuration change needed again to keep things simple.

The single "join enabled" LED is almost a warning that could be referenced in manuals/support docs (i.e: "After you're done setting up, make sure you disable joins. This is indicated with the LED on your coordinator"). The moment there are multiple LEDs blinking/on at the same time (and consider that they are all different colours/locations on the different boards), we are creating a support nightmare for all maintainers involved.

Gio-dot, 2021-03-21:
> As a first step, we can assume that `DIO_7` is present on most boards therefore it could be used as the one LED to be used as the "join enabled" indicator. Perhaps the second LED and configurable-IO [...]

I agree with all your reasoning except for the secon led part: i think that a "zigbee traffic led" is a very useful information and shouldn't confuse users. It will give an instantaneous visual debug info about adapter working behaviour.

omerk, 2021-04-21:
bump, to remove the stale label

Koenkk, 2021-05-22:
LED is now supported in the latest zigbee2mqtt dev branch in combination with the dev firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/develop/coordinator/Z-Stack_3.x.0/bin . Led will be turned on when permit join is enabled.

egony, 2021-08-28:
> Led will be turned on when permit join is enabled.

Koenkk, current led implementation is too dirty - now you are use setLED function in z2m instead of using ZDO_ProcessMgmtPermitJoinReq in firmware. So because of that, true led support are broken now: in my firmware all leds activity controls within firmware - blinking when joining, flash when packet exchange. When permit_join changes, I use **ZDO_ProcessMgmtPermitJoinReq**:

```
  // Blink when Join permited, ON when prohibited 
  if (duration==0) 
  {
    LED_stopBlinking(gGLedHandle);
    if (LEDS_Enabled) LED_setOn(gGLedHandle,100);
  }
  else if (LEDS_Enabled) LED_startBlinking(gGLedHandle, 1000, LED_BLINK_FOREVER);
 ```

When packet received, I use **afIncomingData**:

`
  if (LEDS_Enabled) LED_startBlinking(gRLedHandle, 100, 1);
`

But when setLED toggle in controller.js, firmware just enable or disable whole leds. So used wrong method - in fact you are use disable_led instead permit_join. And it is wrong way.

You have my sources, please, take a look again.

Koenkk, 2021-08-28:
@egony but how to handle the `disable_led` option in that case? (if we don't want to enable the LED when permit join is enabled)

egony, 2021-08-28:
> but how to handle the `disable_led` option in that case?

For disable_led nothing changed. When disable_led=true leds will be torned off forever. In my firmware it works as expected.

egony, 2021-08-28:
> but how to handle the `disable_led` option in that case?

```
  if (mode==0) 
  {
    LEDS_Enabled=false;
    LED_stopBlinking(gGLedHandle);
    LED_stopBlinking(gRLedHandle);
    LED_setOff(gGLedHandle);
    LED_setOff(gRLedHandle);
#ifdef LAUNCHPAD_CONFIG
#else
    LED_stopBlinking(gYLedHandle);
    LED_stopBlinking(gBLedHandle);
    LED_setOff(gYLedHandle);
    LED_setOff(gBLedHandle);
#endif
  }
  else LEDS_Enabled=true;
```
Here

Koenkk, 2021-08-28:
@egony sounds like a good solution, I will incorporate these changes on my next FW iteration, currently waiting for Q3 release of Z-stack firmware from TI.

egony, 2021-08-28:
> Q3 release of Z-stack firmware from TI

Is it not Q3? 
SDK_5.20.00.52 Release Date: 14 Jul 2021

Koenkk, 2021-08-29:
Ah, I mean Q4 (next update)

closed 2021-10-06 by github-actions[bot]